Opentrons/opentrons #9571: bug: protocol with load_labware_from_definition throws movement error during Labware Position Check

hartm9, 2022-02-24:
# Overview

A [customer](https://opentrons.lightning.force.com/lightning/r/Case/5002S00001QjbL3QAJ/view?overrideNavRules=true) sent in a protocol that was throwing a movement error at the beginning of Labware Position Check (see [Confluence page](https://opentrons.atlassian.net/wiki/spaces/technicalsupportwiki/pages/3179479593/Movement+error+after+upgrading+to+5.0.0) for more details, protocol, image of error, case 18515).  This protocol worked in 4.7.0, according to the customer, but was throwing this error after upgrading to 5.0.0.  

When I attempted to load the protocol on my OT-2 and run Labware Position Check, I also got a movement error very similar to the customer's error.  However, when I ran the load_labware_from_definition line in Jupyter notebook, I had no problems, and could move to wells within the two different pieces of custom labware.

I then put the customer's labware definition code (6 lines - definition, json.loads, and load_labware_from_definition commands, for the two pieces of custom labware) into dinosaur.py - which also threw a movement error at the beginning of Labware Position Check

# Steps to reproduce

1. Download protocol 08052021_CAA_H_24 samples_mod.py from [this Confluence page](https://opentrons.atlassian.net/wiki/spaces/technicalsupportwiki/pages/3179479593/Movement+error+after+upgrading+to+5.0.0)
2. Load protocol into Opentrons App, version 5.0.0
3. Click Run Labware Position Check

# Current behavior

The OT-2 throws a movement error at the beginning of Labware Position Check and cannot proceed with the check.

# Expected behavior

The OT-2 can perform Labware Position Check without any movement errors

# Workaround

Build custom labware in Labware Creator and load using load_labware

b-cooper, 2022-05-13:
confirm this is still an issue in edge, if not close

emilywools, 2022-05-13:
al protocols i tried to import failed analysis on edge, so i was not able to test this

mcous, 2022-06-02:
Closed by https://github.com/Opentrons/opentrons/issues/9571

(Was not automatically closed by GitHub because above PR merged into `hotfix_5.0.2` branch rather than `edge`)

PR #9588 (2022-03-01): fix: send custom labware definitions to the server during LPC setup
## Overview

This PR fixes #9571 by allowing the app to explicitly add labware definitions to the `Run` as part of the LPC setup phase. This ensures that any custom labware loaded during the analysis run will be present for LPC load labware commands, even if those labware definitions were specified at runtime in the Python protocol via `load_labware_by_definition`.

## Changelog

- feat(api-client, react-api-client): add utils to create labware definitions
- feat(robot-server): add POST /runs/:id/labware_definitions endpoint
- [... cut]