Opentrons/opentrons #9585: Verify Magnetic Module home-to-base offsets

SyntaxColoring, 2022-02-28:
# Summary

Based on reading some code, some constants used in Magnetic Module height calculations looks like they *might* be wrong.

If they are, this will have implications for PAPIv2, PAPIv3, Protocol Engine, and Protocol Designer: Magnetic Modules may move to the wrong heights sometimes.

# Investigation so far

Magnetic Modules have two "zero positions" that we care about:

1. The home position, which is also the disengage position, and the lowest position that the Magnetic Module can be set to.
2. The labware base position, which puts the tops of the magnets level with the bottom of the labware. This is slightly higher than the home position.

Here is where we define the difference between the two in Python:

https://github.com/Opentrons/opentrons/blob/db96d5ebab82c0b5e22f3607757708b69a80d0c9/api/src/opentrons/hardware_control/modules/magdeck.py#L22-L23

(Note that *I* wrote that comment, and I am not an expert in this code, so that comment could be wrong. See https://github.com/Opentrons/opentrons/pull/9494 for why I think it's right, though.)

To me, this strongly suggests that GEN1 and GEN2 Magnetic Modules are expected to have the same physical home-to-base offset. In other words, *according to this constant,* if you:

1. Home (disengage) a Magnetic Module GEN1 and a Magnetic Module GEN2
2. Record the position of the magnets against a physical reference
3. Gradually raise the magnets until they're level with the labware base plane
4. Record the position of the magnets again
5. Take the difference between the two recorded positions

Then you should get 5 millimeters, for both GEN1 and GEN2. 

However, based on other code, it seems like the offset should be different between GEN1 and GEN2.

When homing, after the magnets hit the limit switch, they back off of it by some constant distance. Where they land after they back off is defined as the home position. That distance, `HOMING_RETRACT`, is [defined as `2` for both GEN1 and GEN2 Magnetic Modules](https://github.com/Opentrons/opentrons-modules/blob/55980fa8440bd40721b30fb87305654ca5725a37/arduino-modules/mag-deck/mag-deck-arduino/motion.h#L9). But @sanni-t pointed out that because GEN1 Magnetic Modules operate in units of half-millimeters, and [`HOMING_RETRACT` is not treated specially](https://github.com/Opentrons/opentrons-modules/blob/55980fa8440bd40721b30fb87305654ca5725a37/arduino-modules/mag-deck/mag-deck-arduino/mag-deck-arduino.ino#L119), that `2` may physically equate to *1 mm* for GEN1 Magnetic Modules instead of 2 mm. This *should* imply a 1 mm difference between the GEN1 and GEN2 home-to-base offsets, but this difference isn't reflected in `OFFSET_TO_LABWARE_BOTTOM`. Unless the 1 mm difference is accounted for by other GEN1/GEN2 hardware differences, or by other code that I'm not seeing, I think this means `OFFSET_TO_LABWARE_BOTTOM` is wrong.

And then we have our JavaScript code, which *does* say there's a difference:

https://github.com/Opentrons/opentrons/blob/add3ff1eafdc98b8e2d1239428e641e49fa39374/shared-data/js/constants.ts#L88-L95

https://github.com/Opentrons/opentrons/blob/add3ff1eafdc98b8e2d1239428e641e49fa39374/shared-data/js/getLabware.ts#L83

But note that these two JavaScript snippets might not even be consistent with each other. The first one says that the GEN1 offset is 5 mm and the GEN2 offset is 4 mm. The second one, I *think,* is indirectly saying that the GEN1 offset is 4 mm, based on its usage and historical context.)

# Acceptance criteria

To close this ticket, we need either:

* Documentation for why this isn't a problem
* If it is a problem, either...
    * A fix for it, taking care to preserve backwards compatibility when necessary, and taking care around the duplication between Python and JavaScript.
    * Further tickets for a fix

SyntaxColoring, 2022-03-03:
Anecdotally, doing:

```python
magnetic_module_gen1.engage(height_from_base=0)
magnetic_module_gen2.engage(height_from_base=0)
```

...and doing a side-by-side comparison and measuring with calipers really makes it look like:

* The magnets on the GEN1 are 1 mm lower than the labware base plane
* The magnets on the GEN2 are properly level with the labware base plane

But take this with a big grain of salt because:

* I'm using sloppy caliper technique
* I think these are weird old QA modules (that we happened to have lying around in the office)
* This GEN1 module's magnets were cannibalized from a different GEN1 unit

mcous, 2022-06-13:
Doing some testing today with a gen1 and a gen2 I was able to find in the office. With these two units

- Gen 1: `5 units` (meaning `2.5 mm`) is `1.5 mm` too low
- Gen 2: `2.5 units` (meaning `2.5 mm`) is correct

Due to lack of available gen1 mag. modules, I'm unable to confirm if the gen1 discrepancy is a systemic issue or a problem with this specific unit

SyntaxColoring, 2022-06-13:
The specific value of 1.5 mm too low is surprising to me. I would have expected 1.0 mm too low, from my investigation above. Could that be measurement error, or were you pretty confident in the 1.5 mm?

mcous, 2022-06-13:
Hard to say. Using my most careful calipering technique, I think I'm actually getting an actual "home to labware base" offset of `3.75 mm`.

I've been tending towards `1.5 mm` because `2.5 + 1.5 = 4.0`, which lines up with the multiple places in the UI code that seem to believe this is the correct value for the "home to labware base" offset (PD and app)

mcous, 2022-06-13:
Proposal:

In the 6.1 release, under the auspices of the PAPI v2.13 bump, fix this offset to our best determinable correct value _and_ change the `module_context.engage` in PAPIv2 to operate in actual mm to align with the app UI, PD, and the ProtocolEnginge.

With that change (and excepting bug preservation logic in PAPIv2), we can move all coordinate conversion to hardware API internals and allow HWAPI clients to operate in real mm across the board.

mcous, 2022-09-26:
Moved to RET-1242

closed 2022-09-26 by mcous