Opentrons/opentrons #9591: feat: Move mount without pipette attached

wdbirch19, 2022-03-01:
# Overview

I only have one pipette mounted to my OT-2 and I would like to be able to move the other mount in the z-axis for additional functionality, however I currently cannot do so through the Python API without a pipette attached. It would be great if this functionality could be added!

mcous, 2022-03-02:
This is an interesting use case! It might line up with some of our internal movement needs for things like pipette movement. I wouldn't expect us to get to this anytime soon, though.

Until that happens, I believe you could do something like this:

```python
from opentrons.types import Mount, Point
from opentrons.protocol_api import ProtocolContext

def run(ctx: ProtocolContext) -> None:
    unsafe_hardware_api = ctx._hw_manager.hardware

    # move a mount without a pipette
    unsafe_hardware_api.move_to(
        mount=Mount.LEFT,                         # < use the left mount
        abs_position=Point(x=..., y=..., z=...),  # < move the bottom of the mount to position
    )
```

**Using `ctx._hw_manager.hardware` is an private, internal implementation detail.** We can't really support using it and we _will_ remove this private interface without warning in a future release

wdbirch19, 2022-03-03:
Hi, thank you very much for getting back to me so quickly - that works great, I'm able to move the additional mount to any x,y,z coordinates. However, when I do so the other mount holding the pipette is retracted (and vice versa). Would it be possible to to move each of the mounts independently from one another so that the other mount doesn't need to home?

For some background: I'm using the robot to perform automatic sampling/dosing of microbial bioreactors housed in the enclosure, and the extra mount will have a custom part attached to hold a sampling port open. So I need the mount to descend and remain in place while the pipette is used for sampling.

Thank you in advance for taking the time to help me with this request.

mcous, 2022-03-04:
> Would it be possible to to move each of the mounts independently from one another so that the other mount doesn't need to home?

Oof, this one is tough. You _might_ be able to do this, but it will be even hackier. Moving one pipette at a time is pretty baked into our system for collision avoidance reasons. You might be able to do something by hacking around with `opentrons.hardware_control.API` internals.

I would take a look at [`API.move_to`](https://github.com/Opentrons/opentrons/blob/64ad11bdb36edf314587c55d2edea3c385b2d507/api/src/opentrons/hardware_control/api.py#L637) and [`API._cache_and_maybe_retract_mount`](https://github.com/Opentrons/opentrons/blob/64ad11bdb36edf314587c55d2edea3c385b2d507/api/src/opentrons/hardware_control/api.py#L704). As with my workaround above, we will are not able to officially support messing around with internals in this way, so please be very careful. There is a relatively high risk of damaged equipment if you start messing with code that prevents collisions 😬

wdbirch19, 2022-03-09:
Thank you once again for your help with this! I fully understand the risks with doing this and appreciate it's way outside of the scope of the robot.

wdbirch19, 2022-03-09:
Thank you once again for your help with this - I fully appreciate the risks involved with changing internal functions in this way and accept that it is entirely my responsibility to avoid collisions etc. If I were to change the move_to function to allow this enable this type of movement, is there a way that I can upload the changed function to the robot, replacing the default files? I understand this is way beyond the scope of what the robot is typically used for

mcous, 2022-03-09:
I _think_ you could get away with this by setting private properties of the `hardware_control.API` object at run time and carefully undoing your changes when you're done.

Otherwise, our [contributing guide](https://github.com/Opentrons/opentrons/blob/edge/CONTRIBUTING.md) details how to set up a local development environment and push changes to your robot, provided you have SSH access set up. Setting up this environment is pretty heavyweight, though, so I would recommend careful runtime patching if you can make it work

wdbirch19, 2022-03-11:
I've been trying to to call the [_move](https://github.com/Opentrons/opentrons/blob/64ad11bdb36edf314587c55d2edea3c385b2d507/api/src/opentrons/hardware_control/api.py#L717) function rather than move_to but am running into issues with defining target_position. Is there a better way of going about this? Many thanks

wdbirch19, 2022-03-28:
I feel like I am very close to getting this to work - having looked through [API.move_to](https://github.com/Opentrons/opentrons/blob/64ad11bdb36edf314587c55d2edea3c385b2d507/api/src/opentrons/hardware_control/api.py#L637) and [API._cache_and_maybe_retract_mount](https://github.com/Opentrons/opentrons/blob/64ad11bdb36edf314587c55d2edea3c385b2d507/api/src/opentrons/hardware_control/api.py#L704) , I believe I could achieve the functionality I need by setting the value of **_last_moved_mount** so that

`if mount != self._last_moved_mount and self._last_moved_mount`

in [API._cache_and_maybe_retract_mount](https://github.com/Opentrons/opentrons/blob/64ad11bdb36edf314587c55d2edea3c385b2d507/api/src/opentrons/hardware_control/api.py#L712) returns **False** and the mount is not homed. However, it looks like I am unable to set this value (I assume it is protected?)... Is there a way around this? I would like to avoid setting up a local environment if possible and this seems like a solution which could work. 

Thank you once again for your time taken to help me with this.