Opentrons/opentrons #9603: bug: OT-2 5.0 or 5.0.1 update completely loose the labware definition to be lost and does not save the calibrations

UfieA, 2022-03-02:
# Overview

# Steps to reproduce

# Current behavior

# Expected behavior

mcous, 2022-03-02:
Hi @UfieA, could you please fill out the pre-existing issue template so that we can better address your issue? Also, please see [our contributing guide](https://github.com/Opentrons/opentrons/blob/edge/CONTRIBUTING.md#opening-issues) for more tips about filing issues.

From reading what you've posted so far, it seems that you are running into a fundamental, and **intentional**, difference between the 4.x software and the 5.x software.

### Version 4.x

1. The user opens a protocols
2. The user measures the difference between the nominal location of a labware's A1 well and its actual location 
3. The robot saves that offset vector to disk, to be automatically applied to **all labware that use that same labware definition**
4. Steps 2 and 3 are repeated for every distinct labware _type_ in the protocol

In our research, we found this behavior:

- Made it too difficult to do the kind of actual fine-tuning needed to ensure proper operation
- Allowed users to "hide" problems with more fundamental calibrations, like the deck calibration or pipette offset calibration, by encoding it into labware offsets

### Version 5.x

1. The user opens a protocol
2. The user measures the difference between the nominal location of a labware's A1 well and its actual location
3. The robot saves the offset vector to memory **for that particular labware of that particular protocol**
4. Steps 2 and 3 are repeated for every single labware _instance_ in the protocol

The reasons we implemented this change are:

- Applying an offset per labware ensures that each labware, even if they share their definition with other labware on the deck, are properly fine tuned
- Scoping offsets to a particular protocol ensures that you always fine tune to a specific equipment setup, so that you're not "papering over" problems with your robot's deck and pipette calibrations

### Robot reboots

> anytime you... turn off and on the system all the labware... [offsets are] erased

"Forgetting" previous runs through reboots is a limitation of the 5.0.x release. In the upcoming 5.1.x release, the robot will be able to remember its previous 20 protocol runs and all the offsets used, so if you re-run your protocol, it will re-use your measured offsets

mcous, 2022-03-02:
Thanks for the extra details @UfieA!

> After updating the app, we ran into the issue of the pipettes consistently being 15mm above the custom labware that we applied

The most likely cause of this problem is that your custom labware definitions themselves are off by about ~15mm, and this was likely "covered over" by the previous labware calibration data in 4.x. Regardless of if you remain on 4.x or upgrade to 5.x, I would recommend that you verify the labware and well position data in the definition itself. 

> Each time we went through the labware offset the system saved it for that run and that run only.

As mentioned in my comment above, this is an intentional change in the 5.x release. If, for example, a custom labware definition is incorrect, the best available fix to a user is to correct the definition.

We try to follow [semantic versioning](https://semver.org/) with our releases, which means that any time we change the _major_ version number (4.x.x > 5.x.x), we have made "breaking" changes that will require changes to user behaviors or workflows.

> the labware offset would try to get us to calibrate the labware with the 300p on 20ul tips

**We're investigating this,** it sounds like a potential bug in the labware position check logic.

> We had spoken to support, and they indicated there should be a folder where the labware settings are saved for each protocol, but there was nothing to be found on either computer

I will follow up with support on this, since I there was a miscommunication here. There is no folder on the user's computer where labware offset data is saved. There is a folder on the user's computer where a user's custom labware **definitions** are saved. Definitions are stored entirely separately from labware offset data in our system.

- 4.x - labware offsets are stored in a folder **on the robot** in `/data`, which you can access via [SSH](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2)
- 5.0 - labware offsets are stored in robot memory. You can access them in the Labware Setup UI in the app
- 5.1+ (future releases) - labware offsets will be stored in a persistent database on the robot. Labware offsets will continue to be tied to specific protocol runs, but the robot will remember your last 20 protocol runs, even through reboots

> We would just like to be able to save the labware somehow and be able to run the assays without doing the labware offset every single time for every single run

In the 5.0 release, this is possible if:

1. You upload the protocol, and
2. You always click the "re-run" protocol in the app, and
3. You keep the OT-2 powered on

In the 5.1 release, condition (3) will no longer apply. If (3) is unacceptable for now, I would recommend waiting on the 4.7 release until 5.1. If (2) is unacceptable, you will need to stay on 4.7. We will not be issuing updates to the 4.x software line, but it your current workflow is stable on that release, it may not matter.

UfieA, 2022-03-02:
Hi Mike,

Thank you for your responses! We tried to go back to 4.7 on the laptop and
the computer after being on the 5.0.1 and neither were able to downgrade
without the issues still remaining. We had updated both the software
version and the robot software version as well. Are we missing something in
the steps for it to be behaving in the same way?

We are thinking of downloading 4.7 on another computer without having had
the software go to 5.0.1 at all and see if that may make a difference.

What are the proper steps to be installing the 4.7 just so we are not
making any mistakes along the way?
What are the proper steps to downgrade to 4.7 from 5.0 or 5.0.1? We had
updated the robot software and then we uninstalled the previous app version
based on the support suggestions- is there something we are missing?

Let me know what you think,

Maggie

mcous, 2022-03-02:
I would recommend:

1. [Uninstall the Opentrons App](https://support.opentrons.com/en/articles/2393514-uninstall-the-opentrons-app)
2. Download the [4.7 installer](https://github.com/Opentrons/opentrons/releases/tag/v4.7.0)
3. Install the 4.7 app. After the app installation completes, verify that you have the correct app version installed by navigating to "More" > "App" > "Software Settings". While here, you probably want to turn the "App Update Alerts" option off.
    [image]
4. Navigate to "Robot" > [Your OT-2] > "Information" and click the "Downgrade" button    
    [image]
5. After the robot has restarted, verify that the the robot's "Information" card says "4.7.0"

> neither were able to downgrade without the issues still remaining

The original post had a few different issues, which _specific_ issues are you observing after downgrading to version 4.7?

mcous, 2022-03-02:
@UfieA the user interface of the "Protocol" page changed drastically between 4.7 and 5.0, so if you aren't noticing a difference then that means you did not successfully downgrade the app.

There are no special instructions for downgrading, you simply have to quit the app and run the 4.7 installer. If you give me more specific information, like the operating system your using and your exact observations at every step of the way, I might be able to help you troubleshoot more.

1. Close the app
2. Follow our [support guide for uninstalling the app](https://support.opentrons.com/en/articles/2393514-uninstall-the-opentrons-app)
3. Complete steps in the "Optional: Remove configuration files" section of the uninstall guide

At this point, make extra certain that the Opentrons App is fully uninstalled. If on Windows, make sure that it isn't in your Start Menu nor `Add/Remove Programs` menu anymore. On `macOS`, make sure that it's not in your `/Applications` nor `~/Applications` folder. After you uninstall, you may want to reboot the computer.

After you're sure the Opentrons App has been uninstalled, proceed with installing the 4.7 release.

5. Download the 4.7 release for your operating system from the [4.7 release page](https://github.com/Opentrons/opentrons/releases/tag/v4.7.0)
6. Install the 4.7 release from the file you downloaded
    - I recommend double checking that you are using the correct file
    - It should have `v4.7.0` and `b13981` in the filename

After the app has been "downgraded" to 4.7, it should report in its "More" > "App" page that it is on `4.7.0`. If it doesn't, you likely have multiple copies of the Opentrons App installed. For example, I believe this can happen on Windows if someone runs the installer once with "Install for all users" and another person runs the same installer with "Install for my user". If this is the case, you may have to get your IT department involved to ensure both copies are removed.

UfieA, 2022-03-03:
>    1. Download the 4.7 release for your operating system from the 4.7
>    2. Install the 4.7 release from the file you downloaded

Hi Mike,

Thank you for sending this information. We are going back to the client to
try it on their computer again today and we will be in touch with you if
there are any issues, but hopefully just with an update that everything is
working smoothly again.

Thank you again and we will be in touch!

Best,

Maggie

pkMyt1, 2022-03-03:
I get the same behavior.  Have to recalibrate for each run and have to calibrate each labware even if they are the same.  For example, if there are two 20 uL tip boxes it requires a calibration for each every time.  I will try a new deck and pipet offset calibration as soon as the run I have going is finished.

mcous, 2022-03-03:
> Have to recalibrate for each run and have to calibrate each labware even if they are the same. For example, if there are two 20 uL tip boxes it requires a calibration for each every time.

@pkMyt1 to be clear, this is a very intentional change of the 5.x. After lots of research, we concluded that "calibrating" by labware type and applying that offset to all future protocols was **not sufficient to ensure proper motion during protocols**, and would lead to "abuse" of the labware offset system by using it to account for errors in the labware definitions themselves.

@UlfieA I'm going to close this ticket, because after investigation the Labware Position Check is currenty acting as we intend it to. We will continue to improve the flow to account for more use-cases, and your feedback has been registered so that we can use it to inform our work going forward.

I have the following recommendations for using v5:

- Do not load equipment (labware, pipettes, modules) that you do not use in the protocol
- If you are applying "large" offsets to custom labware (>5mm) during Labware Position Check, this means your custom labware definition is probably not geometrically correct, and should be fixed

Please feel free to continue posting in this thread, and I will do my best to answer any questions or provide troubleshooting advice!

closed 2022-03-03 by mcous

pkMyt1, 2022-03-03:
> Labware Position Check is currently acting as we intend it to. We will

We always have two pipettes loaded even if a given run only uses one.
 Different protocols might use one or both and are being run one after
another.  Requiring us to remove one each time is not practical and has the
potential for damage.  Nothing is off by more than 1 -2 mm  but even your
predefined tip boxes are off by that amount.  I would have thought those
should be fine.  To be clear, the offset problem is on all axises and is
about the same for all labware, custom or not.

mcous, 2022-03-03:
@pkMyt1 I am **not** suggesting that you physically remove the pipettes. Instead, I'm suggesting that you do not include the `load` commands in your protocol for unused equipment. For example, in a Python protocol, rather than doing this...

```python
def run(ctx):
    # common setup steps, even if not all protocols use this equipment
    tip_rack_300 = ctx.load_labware("opentrons_96_tiprack_300ul", 1)
    tip_rack_20 = ctx.load_labware("opentrons_96_tiprack_300ul", 2)
    p300 = ctx.load_pipette("p300_single_gen2", mount=Mount.LEFT, tip_racks=[tip_rack_300])
    p20 = ctx.load_pipette("p20_single_gen2", mount=Mount.LEFT, tip_racks=[tip_rack_20])

    # ...some protocol that only uses the p300...
```

...you should something like this, where you only issue `load_pipette` and `load_tiprack` to equipment that will actually be used by your protocol...

```python
def setup(ctx, use_p300, use_p20):
    """Common setup function, where you can load only what you need."""
    tip_rack_300 = None
    p300 = None

    tip_rack_20 = None
    p20 = None

    if use_p300:
        tip_rack_300 = ctx.load_labware("opentrons_96_tiprack_300ul", 1)
        p300 = ctx.load_pipette(
            "p300_single_gen2",
            mount=Mount.LEFT,
            tip_racks=[tip_rack_300],
        )

    if use_p20:
        tip_rack_20 = ctx.load_labware("opentrons_96_tiprack_20ul", 2)
        p20 = ctx.load_pipette(
            "p20_single_gen2",
            mount=Mount.RIGHT,
            tip_racks=[tip_rack_20],
        )

    return (p300, tip_rack_300, p20, tip_rack_20)
  

def run(ctx):
    p300, tip_rack_300, p20, tip_rack_20 = setup(ctx, use_p300=True, use_p20=False)

    # ...some protocol that only uses the p300...
```

pkMyt1, 2022-03-03:
Yep, that is how we do it.

UfieA, 2022-03-04:
Hi Mike,

We successfully downgraded to 4.7 on our client's computer and everything
is running smoothly again for them in the lab. We had been trying to use
the update that customer support sent us, and we think some properties may
have been missing because when we took the update directly from Github it
worked flawlessly but would not work on the computer or laptop with what
they had sent.

Thank you for your help- we really appreciate it and it's a relief for our
clients as well.

Have a great weekend!
Maggie

Ceres90, 2022-04-06:
Hi Mike - After reading this thread it seems the labware offset issue we are seeing is not a bug but a feature, just to clarify:  What is the best way to proceed if we have the following setup:   We have two Protocols, Protocol A and Protocol B,  Both protocols have the exact same labware in the exact same layout and use the exact same pipettes. The only difference between them are the pipetting steps (ie pipette 3ul for Protocol A , pipette 5 ul for Protocol B) if i understand this thread correctly, when we switch between protocols we have to redo the labware position check  each time we run the different protocol or is there a way to avoid that?

mcous, 2022-04-06:
@Ceres90 for that particular workflow (switching back and forth between two protocols), I would recommend that you remain on 4.7 until the 5.1 release, which will be our next release. We have improvements planned for both tracking run history of multiple protocols and labware offset sharing that should eliminate needless Labware Position Check runs