Opentrons/opentrons #9617: feat(robot-server): issue stateless control commands to the robot

mcous, 2022-03-03:
## Overview

As a HTTP API user, I would like to command my OT-2 to take simple, stateless actions outside of the context of a protocol. For example, I would like to command an attached temperature module to preheat with a specific target temperture, so that a protocol I run later can take less time.

## Acceptance criteria

- [ ] The robot server has an endpoint that accepts standard ProtocolEngine command requests
    - [ ] A client can request that the server waits to respond until the command is complete
- [ ] The status of any created commands can be queries
- [ ] This endpoint rejects commands if there is a currently active `/run` resource
- [ ] The endpoint can successfully execute the following commands:
    - `home`
    - `setRailLights`
    - `magneticModule/engageMagnet`
    - `magneticModule/disengageMagnet`
    - `temperatureModule/setTargetTemperature`
    - `temperatureModule/deactivate`
    - `thermocycler/setTargetBlockTemperature`
    - `thermocycler/setTargetLidTemperature`
    - `thermocycler/deactivateBlock`
    - `thermocycler/deactivateLid`
    - `thermocycler/openLid`
    - `thermocycler/closeLid`
    - `heaterShakerModule/setTargetTemperature`
    - `heaterShakerModule/setTargetShakeSpeed`
    - `heaterShakerModule/deactivateHeater`
    - `heaterShakerModule/stopShake`
    - `heaterShakerModule/openLatch`
    - `heaterShakerModule/closeLatch`

## Proposed spec

- New endpoints:
    - `GET /commands`
        - Acts the exact same way as `GET /runs/:run_id/commands`
    - `GET /commands/:command_id`
        - Acts the exact same way as `GET /runs/:run_id/commands/:command_id` 
    - `POST /commands`
        - Acts the exact same way as `POST /runs/:run_id/commands` 
- Modified endpoints:
    - `GET /modules`  
        - Module resource has new `id` field to use in the `moduleId` payload fields in `POST /commands` 

## Implementation notes

- This ticket need not concern itself with implementing any commands listed above that are not yet implemented
    - To the extent that it helps, though, we might want to scaffold them out
- I think we should kick the can on pipettes, but at some point we may want stateless pipette commands, too
- Rather than accepting the `protocol_engine.CommandCreate` union, the `/commands` endpoints should spell out the subset of commands that they accept
- The robot-server's `EngineStore` could be augmented to hold on to a "special" direct controls `ProtocolEngine`
    - The engine doesn't have to itself be special, but it could be!
    - We can use existing `EngineStore` conflict logic to prevent clashes between runs and direct controls
- The `ProtocolEngine` that these endpoints use needs to interact with currently attached modules
    - A way to do this might be to put the current modules into the engine state from the route handler prior to creating the command

sanni-t, 2022-03-04:
Will also need to think about modules' firmware update endpoint

y3rsh, 2022-03-04:
What happens when you GET /commands during an active run?

mcous, 2022-03-04:
> Will also need to think about modules' firmware update endpoint

Agreed, but we can probably leave the existing one sitting around for now.

> What happens when you GET /commands during an active run?

Not sure; I'd be down for either a 200 OK with the history (it's a read, so whatever), or a 409 Conflict

commit db0a5268 (2022-03-28): feat(robot-server): allow /commands to receive module commands (#9776)

Closes #9617
closed 2022-03-28 by mcous