Opentrons/opentrons #9646: heater-shaker: user-friendly messages for errors in command responses

sanni-t, 2022-03-07:
# Overview

The heater-shaker module can respond with an error if a specific gcode failed to execute fully. Furthermore, these errors will contain an error code & an error name. The hardware controller can detect these errors and raise, but currently these errors just get raised with the default error from firmware. 

# Implementation details

This ticket aims to improve the current error messaging by either:
1. surfacing these firmware errors with user-friendly messages. So instead of an error saying something like: `err111:main motor:unable to move`, we can raise it with a message 
```
The motor is unable to move following a SetRPM or Home command. Specifically, the motor has not achieved a speed above 20rpm at 1 second after command was sent. Please check for any physical obstruction to the module's plate and try again. If the error persists, contact support.
```
2. Or, if we want to be brief in our error messaging inside the APIs, we can instead create a support document that lists all possible errorr with instructions to address them.

# Acceptance criteria

- If a PAPI/ PD heater-shaker command results in a hardware error, the error is raised & logged correctly by the backend.
- If a heater-shaker HTTP request results in a hardware error, the request gets an error response with an error that a client can show to the user.
- The errors surfaced in both cases should either have error codes + support document to look up error codes OR, the error itself should have an informative message

TamarZanzouri, 2022-06-23:
@laviera after talking to @mcous This ticket is not ready for work yet. @mcous needs to add PR review's to the embedded PR and there needs to be further adjustments. https://github.com/Opentrons/opentrons/pull/10765/files

sanni-t, 2024-02-21:
Addressed as part of the error handling overhaul.

closed 2024-02-21 by sanni-t