Opentrons/opentrons #9648: feat(api): ProtocolEngine-based protocol analysis

mcous, 2022-03-08:
## User story

As an `opentrons` package user, I'd like to analyze my protocols the same way the robot does, so that I can identify expected equipment and the anticipated run log. I'd like this information to be available in both human and machine readable formats

## Acceptance criteria

Once `opentrons` is installed from PyPI, a caller may:

- [ ] Use `opentrons analyze <files...>` to get a human-readable protocol analysis
- [ ] Use `opentrons analyze --json <files...>` to get a machine-readable (JSON) analysis

## Implementation notes

- The [click](https://github.com/pallets/click) package is already on the OT-2, we should use it

### Proof of concept

I've created a proof-of-concept for this CLI: https://github.com/Opentrons/opentrons/compare/edge...api_new-analyzer-proof-of-concept. To try it out, pull down the branch and run `make -C api setup`

#### Human mode

```shell
❯ pipenv run opentrons analyze ./tests/opentrons/data/mad_mag_v2.py
/Users/mc/.opentrons/robot_settings.json not found. Loading defaults
/Users/mc/.opentrons/deck_calibration.json not found. Loading defaults

Files
================================================================================
Main:     mad_mag_v2.py
--------------------------------------------------------------------------------

Modules
================================================================================
Slot 4:   magneticModuleV1
--------------------------------------------------------------------------------

Labware
================================================================================
Slot 12:  opentrons_1_trash_1100ml_fixed
Slot 1:   opentrons_96_tiprack_300ul
Slot 4:   nest_96_wellplate_100ul_pcr_full_skirt on magneticModuleV1
--------------------------------------------------------------------------------

Pipettes
================================================================================
Right:    p300_single_gen2
--------------------------------------------------------------------------------

Commands
================================================================================
1:        Load labware opentrons_96_tiprack_300ul in slot 1
2:        Load module magneticModuleV1 in slot 4
3:        Load labware nest_96_wellplate_100ul_pcr_full_skirt on magneticModuleV1 in slot 4
4:        Load pipette p300_single_gen2 on right mount
5:        Disengaging Magnetic Module
6:        Pick up tip from A1 of opentrons_96_tiprack_300ul in slot 1 with right p300_single_gen2
7:        Aspirating 30.0 uL from A1 of NEST 96 Well Plate 100 µL PCR Full Skirt on Magnetic Module on 4 at 46.43 uL/sec
8:        Dispensing 30.0 uL into B1 of NEST 96 Well Plate 100 µL PCR Full Skirt on Magnetic Module on 4 at 46.43 uL/sec
9:        Drop tip into A1 of opentrons_1_trash_1100ml_fixed in slot 12 with right p300_single_gen2
10:       Engaging Magnetic Module
11:       Disengaging Magnetic Module
12:       Engaging Magnetic Module
13:       Disengaging Magnetic Module
14:       Engaging Magnetic Module
15:       Disengaging Magnetic Module
16:       Engaging Magnetic Module
17:       Disengaging Magnetic Module
--------------------------------------------------------------------------------
```

#### Machine mode

```shell
❯ pipenv run opentrons analyze --json ./tests/opentrons/data/mad_mag_v2.py
/Users/mc/.opentrons/robot_settings.json not found. Loading defaults
/Users/mc/.opentrons/deck_calibration.json not found. Loading defaults
{"commands": [{"id": "commands.LOAD_LABWARE-0", "createdAt": "2022-03-08T13:46:16.134776+00:00", "commandType": "loadLabware", "key": "commands.LOAD_LABWARE-0", "status": "succeeded", "params": {"location": {"slotName": "1"}, "loadName": "opentrons_96_tiprack_300ul", "namespace": "opentrons", "version": 1, "labwareId": null, "displayName": null}, "result": {"labwareId": "labware-0", "definition": {"schemaVersion": 2, "version": 1, "namespace": "opentrons", "metadata": {"displayName": "Opentrons 96 Tip Rack 300 \u00b5L", "displayCategory": "tipRack", "displayVolumeUnits": "\u00b5L", "tags": []}, "brand": {"brand": "Opentrons", "brandId": [], "links": ["https://shop.opentrons.com/collections/opentrons-tips/products/opentrons-300ul-tips"]}, "parameters": {"format": "96Standard", "quirks": null, "isTiprack": true, "tipLength": 59.3, "tipOverlap": 7.47, "loadName": "opentrons_96_tiprack_300ul", "isMagneticModuleCompatible": false, "magneticModuleEngageHeight": null}, "ordering": [["A1", "B1", "C1", "D1", "E1", "F1", "G1", "H1"], ["A2", "B2", "C2", "D2", "E2", "F2", "G2", "H2"], ["A3", "B3", "C3", "D3", "E3", "F3", "G3", "H3"], ["A4", "B4", "C4", "D4", "E4", "F4", "G4", "H4"], ["A5", "B5", "C5", "D5", "E5", "F5", "G5", "H5"], ["A6", "B6", "C6", "D6", "E6", "F6", "G6", "H6"], ["A7", "B7", "C7", "D7", "E7", "F7", "G7", "H7"], ["A8", "B8", "C8", "D8", "E8", "F8", "G8", "H8"], ["A9", "B9", "C9", "D9", "E9", "F9", "G9", "H9"], ["A10", "B10", "C10", "D10", "E10", "F10", "G10", "H10"], ["A11", "B11", "C11", "D11", "E11", "F11", "G11", "H11"], ["A12", "B12", "C12", "D12", "E12", "F12", "G12", "H12"]], "cornerOffsetFromSlot": {"x": 0, "y": 0, "z": 0}, "dimensions": {"yDimension": 85.48, "zDimension": 64.49, "xDimension": 127.76}, "wells": {"A1": {"depth": 59.3, "x": 14.38, "y": 74.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "B1": {"depth": 59.3, "x": 14.38, "y": 65.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "C1": {"depth": 59.3, "x": 14.38, "y": 56.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "D1": {"depth": 59.3, "x": 14.38, "y": 47.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "E1": {"depth": 59.3, "x": 14.38, "y": 38.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "F1": {"depth": 59.3, "x": 14.38, "y": 29.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "G1": {"depth": 59.3, "x": 14.38, "y": 20.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "H1": {"depth": 59.3, "x": 14.38, "y": 11.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "A2": {"depth": 59.3, "x": 23.38, "y": 74.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "B2": {"depth": 59.3, "x": 23.38, "y": 65.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "C2": {"depth": 59.3, "x": 23.38, "y": 56.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "D2": {"depth": 59.3, "x": 23.38, "y": 47.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "E2": {"depth": 59.3, "x": 23.38, "y": 38.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "F2": {"depth": 59.3, "x": 23.38, "y": 29.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "G2": {"depth": 59.3, "x": 23.38, "y": 20.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "H2": {"depth": 59.3, "x": 23.38, "y": 11.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "A3": {"depth": 59.3, "x": 32.38, "y": 74.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "B3": {"depth": 59.3, "x": 32.38, "y": 65.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "C3": {"depth": 59.3, "x": 32.38, "y": 56.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "D3": {"depth": 59.3, "x": 32.38, "y": 47.24, "z": 5.39, "totalLiquidVolume": 300, "xDimension": null, "yDimension": null, "diameter": 5.23, "shape": "circular"}, "E3": {"depth": 59.3, "x": 32.38 [... cut ...] magneticModuleV1", "location": {"slotName": "4"}, "definition": {"otSharedSchema": "module/schemas/2", "moduleType": "magneticModuleType", "model": "magneticModuleV1", "labwareOffset": {"x": 0.125, "y": -0.125, "z": 82.25}, "dimensions": {"bareOverallHeight": 110.152, "overLabwareHeight": 4.052, "lidHeight": null}, "calibrationPoint": {"x": 124.875, "y": 2.75}, "displayName": "Magnetic Module GEN1", "quirks": [], "slotTransforms": {}, "compatibleWith": []}, "serialNumber": "dummySerialMD"}]}
```

SyntaxColoring, 2022-03-08:
I think we should stick with the existing `opentrons_simulate` command instead of moving stuff under a monolithic top-level `opentrons` command.

* The `opentrons` package is already doing a bit too much, and I think a top-level `opentrons` command would make that harder to fix. For example, `opentrons_simulate` and `opentrons_execute` have different dependences, different system requirements, and different audiences, so it would be nice to split them into different packages, someday. 
* There's a lot of user and support momentum behind `opentrons_simulate`.

mcous, 2022-03-10:
If we were to move to an `opentrons` command, I don't think we should remove `opentrons_simulate` and `opentrons_execute`. At most, whenever we thought `opentrons` was ready, we'd point the old ones to the new ones.

(That being said, I think `opentrons_execute` is a bit of an outlier and probably shouldn't be part of the pip package at all)

I have always found `opentrons_simulate` confusing. As a user, when I `pip install` a package with a CLI component, I fully expect the CLI command to share its name with the package. I think anything else is a bit user-hostile.

True story: every time I go to test our CLI  (which I do with often to help out with release QA), I end up doing:

1. `pip install opentrons`
2. `opentrons --version`
3. `zsh: command not found: opentrons`
4. "Oh 💩, the release is broken!"
5. "Oh right, the command is `opentrons_simulate` 🤦"

SyntaxColoring, 2022-03-10:
That makes sense.

I guess I've always thought it was the *package* name that we should change to resolve that inconsistency, not the *command* name. That is, I'd prefer for users to do `pip install opentrons-simulate` and run `opentrons_simulate`. (And `opentrons_execute` should be in another package, as you said.) Basically just because plain `opentrons` is such a big, important name, and I think I'm uncomfortable "using it up."

But that comes down to style; I'm fine with an `opentrons` top-level command if we still preserve `opentrons_simulate`.

PR #9862 (2022-04-01): feat(api): add ProtocolEngine-based analysis CLI
## Overview

Closes #9648

## Changelog

- feat(api): add ProtocolEngine-based analysis CLI

## Review requests

Usage:

```shell
# in the api directory
cd api

# show help
pipenv run python -m opentrons.cli analyze --help

# analyze a protocol
pipenv run python -m opentrons.cli analyze --json path/to/protocol_file_or_dir
```

## Risk assessment

Low. Additionally, not yet exposed to users via entrypoints
closed 2022-04-01 by mcous