Opentrons/opentrons #9655: Create endpoint for GET and POST of robot name

mattwelch, 2022-03-08:
As a robot client, I would like to be able to observe and change the name of the robot, without impacting other robot identifiers (such as the serial number).

## Acceptance Criteria
- [ ] A new (or existing, brought up to date) endpoint exists that supports GET to read robot name and POST to change it
- [ ] No other robot identifiers are impacted

mattwelch, 2022-03-08:
Hey team! Please [add your planning poker estimate](https://app.zenhub.com/workspaces/opentrons-cpx-5e1f6ed55b295a3f514f2283/issues/Opentrons/opentrons/9655?planning-poker) with ZenHub @mcous @sanni-t @SyntaxColoring @TamarZanzouri @y3rsh

SyntaxColoring, 2022-03-14:
# Background

> * [ ] No other robot identifiers are impacted

As far as I know, there are 3 separate "names" that we might need to care about here:

* **Serial number**
    * Intended to be stable and set at the factory.
    * Stored in `/etc/serial`.
    * ⚠️ Caveat: Since the OT-2 has no EEPROM, this gets nuked whenever users replace the SD card. (It default to something generic like `opentrons`, I think.)
    * ⚠️ Caveat: For a large body of robots, this was apparently never set to a true serial number. Instead, it will contain something like `muddy-silence` instead.
    * ⚠️ Caveat: Some users who got assistance from Support to rewrite their robot's *display name* have also rewritten their robot's serial number to whatever they were setting the display name to. Support has two separate scripts for these actions, but they're easy to conflate, and have definitely been conflated before.
* **Hostname**
    * Lets you `ping` or `ssh` or `curl` `hostname.local`.
    * Shows up in troubleshooting logs.
    * That's it. The hostname has no effects that face "normal" users, as far as I know.
    * Normally, it's derived automatically from the serial number. ⚠️ So, it shares all all of the serial number's caveats.
* **Display name (aka "name", aka "pretty hostname")**
    * What shows up in the Opentrons App.
    * Stored in `/etc/machine-info`.
    * Used as the mDNS *service name, which is different from the *hostname*. I don't know what this means or whether it has any meaningful implications.
    * ⚠️ Note: On some OT-2s (the pre-refresh ones?) the *display name* is set to the serial number, and the *serial number* is set to something bogus like `muddy-silence`. It's unclear to me how many robots are like this. We *might* be able to estimate the number from remote logs.

(All of this is partially based on anecdotal Support experience, partially based on fiddling with the robots I've had access to, and partially based on reading `opentrons/update-server/otupdate/common/name_management.py`.)

# Proposal

* We keep the existing `GET`/`POST` `/server/name` endpoints. As far as I know, these, correctly, set the display name and nothing else.
* That's it. **We do nothing specifically to "rescue" the value of the serial number.** In other words, if `/var/serial` is bogus, and the user overwrites the display name which formerly contained the only record of the serial number, we just let that happen. We allow the robot to be left with no correct record of its true serial number.
* Ideally, for bonus points, we add an HTTP endpoint to explicitly overwrite the serial number.
    * This would be nice for correcting bad serial numbers from the aforementioned provisioning bugs, and from fresh SD cards replacements.
    
My reasoning, in short:

* The true serial number is always physically printed on a sticker on the OT-2, if someone really needs it.
* There is absolutely no trustworthy digital record of an OT-2's serial number. (For why not, see all the caveats above.)
* Because of this, any kind of "preservation" or "rescue" logic that we implement in software will be subtle, confusing, and unavoidably wrong at least some of the time. Garbage in, garbage out. So we'd be adding another layer of naming weirdness.
* Many robots *already* operate without any correct record of their serial number, and it's not the end of the world.
    * If we want to change this, we should be talking about serial number *correction,* not *preservation.* The "bonus points" part of my proposal does this.
* If what we need is a trustworthy stable identifier (to track robots across renames, for example), we can pursue other technical options that are totally decoupled from the serial number.
* If, after all of this, we really still want to "preserve" or "rescue" serial numbers, the client can do that with two separate `/name` and `/serial` endpoints. It's a clearer API for this logic to be on the client and for these two endpoints to be separate, anyway.

Happy to elaborate on any of this, if needed.

SyntaxColoring, 2022-03-15:
> Are the current `GET/POST /server/name` endpoints consistent with our "new" endpoints? Are they in the same part of the codebase? In other words, is there any work that needs to be done to make this part of what we're considering our modern 1st and 3rd party dev platform?

Off the top of my head: we happen to serve this endpoint from update-server, not robot-server, and that causes a couple of oddities:

1. The URL `/server/name`. `/server/` is only in there so our system knows to route requests to the update-server process instead of robot-server. It's a leaky abstraction from the perspective of the HTTP client.
2. It doesn't show up in our autogenerated OpenAPI spec.

mattwelch, 2022-03-17:
Closing in favor of using existing endpoints.