Opentrons/opentrons #9670: bug: Hitting cancel run during the end of LPC while robot is homing

nusrat813, 2022-03-09:
[image]

# Steps to reproduce

1. Go through LPC flow
2. Apply offsets 
3. The robot should be homing 
4. Hit `cancel run`

anthonywilleee, 2022-04-15:
investigate to make sure this doesnt happen with new UI

mikecardarelli, 2022-06-03:
@hartm9 to check if this is still happening

hartm9, 2022-06-03:
Still seeing this behavior.  The Cancel Run button is blocked by the "Robot is in motion" pop up while the tip is being returned after LPC, but when the homing process starts, the "Robot is in motion" pop up disappears.  If I click through the "apply labware offsets" pop up and try to hit cancel, I get the following screen that doesn't resolve:
[image]

I've been waiting for a couple of minutes so far, and no change.  I'm able to click around to different tabs, etc., but the run status still doesn't change.

b-cooper, 2022-06-07:
Closing as this is now fixed on edge. This dead-end does not appear to be reachable since #10170 was merged.

closed 2022-06-07 by b-cooper