Opentrons/opentrons #9691: App:  Labware Offsets are all applied with the same pipette

alexjoel42, 2022-03-14:
# Overview

The Ot-2 should not calibrate all tip racks with the same pipette. 
[image]

# Steps to reproduce

1. Load protocol
2. Try to do Labware Offset
3. All Labware Offsets are done by the same pipette 

```Python3
from opentrons import protocol_api

 

# metadata

metadata = {

    'protocolName': 'My Protocol',

    'apiLevel': '2.11'
}

 

# protocol run function. the part after the colon lets your editor know

# where to look for autocomplete suggestions

def run(protocol: protocol_api.ProtocolContext):

    # labware

      quantPlate = protocol.load_labware('biorad_96_wellplate_200ul_pcr', 4)

      storePlate = protocol.load_labware('biorad_96_wellplate_200ul_pcr', 1)

      reagentRack = protocol.load_labware('opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical', 9)

      tiprackA = protocol.load_labware('opentrons_96_tiprack_300ul', '2')

      tiprackB = protocol.load_labware('opentrons_96_tiprack_1000ul', '8')

 

    # pipettes

      left_pipette = protocol.load_instrument('p300_multi_gen2', 'left', tip_racks=[tiprackA])

      right_pipette = protocol.load_instrument('p1000_single_gen2', 'right', tip_racks=[tiprackB])

     

 

    # commands

      right_pipette.distribute(500, reagentRack.wells_by_name()['A3'], quantPlate.wells()[:])

```

# Current behavior
All Labware Offsets are done with the right channel pipette

# Expected behavior

The initial tip rack is calibrated with the right pipette, the rest of the calibrations are done with the left pipette.

alexjoel42, 2022-03-14:
Work around: 

Both pipettes need to be used in the protocol, or any unused pipette should be commented out. Otherwise, one pipette is used to calibrate all tip racks. This can lead to a p1000 trying to pick up 300uL tips (like in this case).

alexjoel42, 2022-03-16:
I just wanted to add some use cases, because I think that the fundamental assumption is "customers won't define unused labware on the deck.

Here are a few use cases (based on customer experiences) that I have seen:

1. Folks can have one "basic protocol"  that they modify for different locations. 
2. 

3.

Matt-Zwimpfer, 2022-03-23:
@nusrat813 to review

PR #10295 (2022-05-17): fix(app): do not include tipracks in Labware Position Check that are unused in protocol
# Overview

Filter out all tipracks that are loaded but unused when constructing the Labware Position Check
Steps. This avoids the possible edge cases where unused and incompatible tipracks are potentially
used by the primary pipette for tip pick in LPC.

Close #9691

# Review requests

Upload a protocol that loads, but doesn't use a tip rack. Run LPC, which should not include that tip rack.

# Risk assessment
low
commit 80eea2b7 (2022-05-17): fix(app): do not include tipracks in Labware Position Check that are unused in protocol (#10295)

Filter out all tipracks that are loaded but unused when constructing the Labware Position Check
Steps. This avoids the possible edge cases where unused and incompatible tipracks are potentially
used by the primary pipette for tip pick in LPC.

Close #9691
closed 2022-05-17 by b-cooper