Opentrons/opentrons #9800: shared-data: add modules' hardware constraints to module definitions

sanni-t, 2022-03-28:
Each module has certain hardware-defined constraints like min or max temperature. We should add these constraints to the module definitions so they can be easily accessed as a single source of truth from various API/ engine components as well as app components. 
Properties to be added to the different module defs:

### Temperature-module
- Min/ Max temperature

### Magnetic module
- Max engage height

### Thermocycler module
- Min/ Max block temperature
- Min/ Max lid temperature

### Heater-Shaker module
- Min/ Max plate temperature
- Min/ Max shake speed

jbleon95, 2022-05-27:
@sanni-t @mcous Do these need to be added to both the `shared_data/module/definitions/2` and `shared_data/module/definitions/3` json definitions are just 3?

And as far as format, I was thinking of adding a top level `constraints` key, with a dictionary of the specific min/max values as the value.

mcous, 2022-06-09:
Closed by #10532

closed 2022-06-09 by mcous