Opentrons/opentrons #9832: bug(PAPIv2): distribute silently no-ops if aspirate volume + disposal volume exceeds tip/pipette max

mcous, 2022-03-30:
## Overview

If the input volume and the input or default disposal volume exceed the pipette or tip's maximum volume, a `distribute` command will silently "no-op" instead of erroring or proceeding.

The "no-op" consists of a tip pickup followed by an immediate tip trash.

## Steps to reproduce

Run the following in `opentrons_simulate`:

```python
from opentrons import protocol_api

metadata = {
    "protocolName": "Distribute no-op",
    "apiLevel": "2.12",
}

def run(protocol: protocol_api.ProtocolContext):
    tip_rack_1 = protocol.load_labware("opentrons_96_filtertiprack_200ul", "3")
    source = protocol.load_labware("opentrons_24_tuberack_nest_1.5ml_snapcap", "1")
    dest = protocol.load_labware("corning_96_wellplate_360ul_flat", "4")
    p300_single = protocol.load_instrument(
        "p300_single_gen2", "right", tip_racks=[tip_rack_1]
    )

    p300_single.distribute(200, source["A3"], dest.columns("9"))
```

Result:

```
Distributing 200.0 from A3 of Opentrons 24 Tube Rack with NEST 1.5 mL Snapcap on 1 to A9 of Corning 96 Well Plate 360 µL Flat on 4
	Transferring 200.0 from A3 of Opentrons 24 Tube Rack with NEST 1.5 mL Snapcap on 1 to A9 of Corning 96 Well Plate 360 µL Flat on 4
		Picking up tip from A1 of Opentrons 96 Filter Tip Rack 200 µL on 3
		Dropping tip into A1 of Opentrons Fixed Trash on 12
```

## Workarounds

- Use `transfer`, not `distribute`
- Pass `disposal_volume=0` to `distribute`

mcous, 2023-09-13:
@ecormany is this one tracked in Jira? I'd love to get this issue off my GitHub list so it can be a little more useful for me!

ecormany, 2023-09-13:
Thanks for the cleanup efforts, Mike!

Moved to RSS-344.

closed 2023-09-13 by ecormany