Opentrons/opentrons #9851: Increasing font size and a number of small changes

mmencarelli, 2022-03-31:
## Overview

Looking to increase the size of a few type styles to make the app more readable. This affects a few components. 

## Acceptance Critera
That the app matches the design in the following ways: 

**Type changes**
- [ ] h1 is now 19/24px, 1.88rem/1.5rem
- [ ] P style is now 13/20px, .813rem/1.25rem
- [ ] our label is now 11px/12px, .688rem/.75rem

**De-scoping**
- [ ] Remove module icon column on the protocol cards

**Mini toast style**
- [ ] Add new mini toast style (view on [figma](https://www.figma.com/file/l7BAJAGICr10ELsSqQD0Sr/UI-Overhaul?node-id=2327%3A85831))
- [ ] Make sure the min-height on the mini toast card is 32px

**Module cards**
- [ ] Change the size of the img of the module to 60x54px
- [ ] Change the 'firmware update' toast in all of the module cards to be the new mini toast style
- [ ] Change the h/s 'module is too hot' toast in to be the new mini toast style
- [ ] Change the pipette 'POC missing' toast in to be the new mini toast style
- [ ] Change the pipette 'POC needed' toast in to be the new mini toast style
- [ ] Change the pipette 'POC recommended' toast in to be the new mini toast style

koji, 2022-04-14:
Type changes
The second and the third were made by brent's PR #9951

mikecardarelli, 2022-05-03:
Mel to split this ticket up

mmencarelli, 2022-05-11:
@mikecardarelli Are you able to close this ticket? I've pulled the outstanding items and merged them with future tickets

closed 2022-05-17 by mikecardarelli