Opentrons/opentrons #9861: Robot Settings: pipette offset calibration or tip length calibration history page

skyewindsor, 2022-04-01:
# Overview

There are calibration history pages that need to be added to the app. These pages can be accessed on the Robot Settings page when a user clicks either of the "See all Pipette Offset Calibration history" or "See all Tip Length Calibration history" ctas. 

# Implementation details

Each row of the calibration history rows includes an overflow menu with the following options and logic. There is also an "export list to .csv" cta at the top of the page.
[image]

# Design

https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx?node-id=1297:90420#172080295

# Acceptance criteria
- [ ] The Pipette Offset Calibration history page is launched when a user clicks "See all pipette offset calibration history"
- [ ] The Tip Length Calibration history page is launched when a user clicks "See all Tip Length Calibration history"
- [ ] Overflow menu greys out "Recalibrate Pipette Offset" if pipette unattached
- [ ] Overflow menu greys out "Recalibrate Tip Length and Pipette Offset" if pipette not attached
- [ ] Export list to .csv downloads csv to local drive

skyewindsor, 2022-04-04:
Descoped from 5.1

koji, 2023-02-27:
The ticket that is related to this is in Jira and edge has been updated

closed 2023-02-27 by koji