Opentrons/opentrons #9877: LPC bug:

skyewindsor, 2022-04-04:
# Overview

In 5.0, if LPC is completed and a protocol is started, but not completed, the LPC data is not saved. If a user partially completes a run and needs to restart, their LPC offsets are not saved. This seems to counter the goal of being able to re-run a protocol with the same offsets.

# Steps to reproduce

Load a protocol, complete LPC, start the protocol, and then cancel the run. LPC data will not be saved. 

# Current behavior

LPC data is not saved when a protocol run is cancelled. 

# Expected behavior

The LPC offsets should be saved to the app, even if the protocol run is cancelled.

mikecardarelli, 2022-05-03:
This should be fixed by LPC enhancements in 5.1

b-cooper, 2022-05-13:
no longer a problem in edge

closed 2022-05-13 by b-cooper