Opentrons/opentrons #9890: LPC bug: incorrect offsets for protocols with single and multi channel pipettes

skyewindsor, 2022-04-05:
# Overview

In protocols with a single and multi channel pipette - there are issues with offsets. For example - if a P20S and P20M are mounted and go to a plate on the deck, based on the above logic, only the P20S will calibrate to the plate. But when the run executes, the P20S calibration looks good, but the P20M is off in the X or Y. This causes issues specifically with the 384 well plate and the P20M and there can be collisions.

Offset issue also applies to a P300M going to a 96-well plate.

In past versions - we were able to run a mock protocol with the P20M and the 384 plate, save the labware offset, then run our “actual” protocol. But now, the labware offsets are not saved - so our workaround for not having a multichannel as part of the LPC flow doesn’t work.

More details from this doc: https://opentrons.atlassian.net/wiki/spaces/technicalsupportwiki/pages/edit-v2/3210215543

# Steps to reproduce

Create a protocol with a single channel pipette, multichannel pipette and 96 well plate or 384 well plate
Execute LPC
Run the protocol 
Observe interaction between multichannel pipette and 96 or 384 well plate

# Current behavior

Multichannel pipette offsets are not saved correctly during LPC, which is causing labware collisions with both 384 and 96 well plates. 

# Expected behavior

Multichannel pipettes should be able to interact successfully with 96 well plates and 384 well plates whether or not a single channel pipette is also attached.

mikecardarelli, 2022-06-28:
@skyewindsor to attempt to repro

skyewindsor, 2022-06-28:
I was not able to reproduce a collision with either a P20M + 24 channel reservoir (similar dims to 382 well plate) or P300M + 96 well pcr plate. 

I did visually see that the P20M had a significant X-offset during the protocol though, which could cause 384 well collisions. LPC was successfully completed with a P1000S on the same labware. 

I say we put this one back in the backlog to address post-6.0.

mikecardarelli, 2022-07-08:
Looks like a calibration issue on the user's robot. Closing this one

closed 2022-07-08 by mikecardarelli