Opentrons/opentrons #9912: bug: Version 5.0.0+ Opentrons App fails "unexpectedly" on some Mac computers

Matt-Zwimpfer, 2022-04-07:
# Overview
There have been several user cases where they try to use the Opentrons App on a Mac, and are even able to secure a connection with the robot for a few minutes, but then the Opentrons App crashes and sends the following error message:

[image]

This occurs with or without a connection to a robot (opening the App on its own) and has been observed on a range of MacOS versions running Opentrons App 5.0.0 - 5.0.2 software versions. 

# Steps to reproduce

1. Have a Mac computer with an Opentrons App version 5.0.0 or above
2. Open up the App and begin to navigate through it (need to just wait until the error occurs)
3. After a few minutes, the App may crash and throw the error message shown above

*Will continue to update this reproducibility as similarities between user cases are examined and confirmed*

# Current behavior
MacOS running Opentrons App on 5.0.0+ may quit unexpectedly and prevent the user from using the App, with or without a robot.

# Expected behavior
Opentrons App should not quit unexpectedly

mcous, 2022-04-07:
This will be difficult to resolve, due to the likelihood that it's something deep in Electron, which we do not control. Things that would be helpful to collect from crashes:

- Exact operating system version
- Device architecture (x64 or ARM)
- App logs (Help > View Opentrons App Logs)

If there's nothing helpful in there, I'd operate on the assumption that this is a crash due to Election, and recommend we work on the following two items:

- Upgrade to a newer version of Electron
- (If crashes are from M1 macs) Start publishing ARM-native app builds

mcous, 2022-04-11:
A "full" uninstall (app + config files) has not resolved the issue for one user. There is nothing untoward in the logs from this installation (e.g. uncaught exception or similar), suggesting that our best path forward on this is to upgrade the underlying Electron framework to a newer release

closed 2025-12-15 by sfoster1