Opentrons/opentrons #9922: Add 4 new labware definitions to labware dropdown in PD

skyewindsor, 2022-04-08:
# Overview

The [four new additions](https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/8649) to the Labware Library need to be added to Protocol Designer. 

# Implementation details

Labware Library links to each of the four new definitions: 

[Applied Biosystems MicroAmp 384 Well Plate 40 µL](https://labware.opentrons.com/appliedbiosystemsmicroamp_384_wellplate_40ul?category=wellPlate)
[Bio-Rad 384 Well Plate 50 µL](https://labware.opentrons.com/biorad_384_wellplate_50ul?category=wellPlate)
[Thermo Scientific Nunc 96 Well Plate 1300 µL](https://labware.opentrons.com/thermoscientificnunc_96_wellplate_1300ul?category=wellPlate)
[Thermo Scientific Nunc 96 Well Plate 2000 µL](https://labware.opentrons.com/thermoscientificnunc_96_wellplate_2000ul?category=wellPlate)

# Design

New plates should show up in the "add labware" dropdown list in PD: 

[image]

# Acceptance criteria

- [ ] Applied Biosystems MicroAmp 384 Well Plate 40 µL shows up in "add labware" dropdown list in PD
- [ ] Bio-Rad 384 Well Plate 50 µL shows up in "add labware" dropdown list in PD
- [ ] Thermo Scientific Nunc 96 Well Plate 1300 µL shows up in "add labware" dropdown list in PD
- [ ] Thermo Scientific Nunc 96 Well Plate 2000 µL shows up in "add labware" dropdown list in PD
- [ ] All 4 additions have appropriate slideout hover state in the drop down list
- [ ] All 4 additions can be added to any deck slot in PD
- [ ] The 2 Thermo definitions can be added to the Magnetic Module in PD ([dependent on this ticket](https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/9921))

skyewindsor, 2022-04-15:
Skye to track and test once new PD release goes out

skyewindsor, 2022-06-28:
@sakibh could this fold in to the PD liquid setup work you are doing this week?

mikecardarelli, 2022-09-09:
Closing this ticket as a duplicate of this ticket: https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/9878

closed 2022-09-09 by mikecardarelli