Opentrons/opentrons #9927: feat: Development environment with nix-shell as alternative to DEV_SETUP.md

barakplasma, 2022-04-11:
# Overview

The `DEV_SETUP.md` is quite complicated, with many manual steps, especially if someone only wants to work on part of opentrons. This feature request (which I could make a PR for) would be to add a development environment with nix-shell as an alternative.

# Implementation details

A section of DEV_SETUP.md or a replacement would be:

1. Install Nix (or use Nix docker)
    1. Install docker, then run `docker run -it -p 8080:8080 -v $(pwd)/:/workdir nixos/nix` from opentrons mono-repo
1. run `cd workdir` then `nix-shell`
1. use this file `shell.nix`
```nix
{ pkgs ? import (fetchTarball "https://github.com/NixOS/nixpkgs/archive/715dc137b08213aabbbe0965b78ab938e5d8d3b7.tar.gz") {}
}:

pkgs.mkShell {
  buildInputs = [
    pkgs.nodejs-14_x
    pkgs.yarn
    pkgs.libudev0-shim
    pkgs.python37
    pkgs.gnumake
    pkgs.curl
    pkgs.openssh
    pkgs.git
  ];
}
```
the development environment would be ready, leaving a dev to just run
`cd protocol-designer; make dev`

and see it just work

# Design

N/A

# Acceptance criteria

The `shell.nix` file above let's me hack on the protocol designer already; but it doesn't support the root `make setup` command yet due to lack of support for the `usb-detection` npm package so far.

mcous, 2022-04-11:
Hi @barakplasm, this is an interesting proposal! While, I don't think we have the resources to test and support an "officially blessed" Docker/NixOS-based development environment, if you were to create a GitHub repository or Gist with your `nix-shell` configuration + setup instructions, I think that would be fantastic. I'd definitely be down to link to those instructions under an "Alternative Community Setups" section or something similar

barakplasma, 2022-04-11:
Sure, I can make an external repo for this contribution

barakplasma, 2022-04-26:
@mcous I opened a repo here with instructions on how to use nix-shell for dev setup (at least on the protocol designer).
I'll make a PR after I use this alternate setup a bit longer