Opentrons/opentrons #9965: HTTP API: separate protocol analysis endpoint

mcous, 2022-04-14:
## Overview

I really thought this ticket already existed, but I couldn't find it. Please close if this turns out to be a duplicate.

As an HTTP API user, I would like the `GET /protocols` endpoint to respond quickly. Analysis results tend to be very large and cause the response to take a long time.

## Acceptance criteria

- `GET /protocols` and `GET /protocols/:id` does not provide full analysis responses, but does:
    - Provide enough information to tell whether a protocol has a completed analysis
- `GET /protocols/:id/analyses` returns full analyses

sanni-t, 2022-04-14:
> `GET /protocols/:id/analyses` returns full analyses

Should this be `/protocols/:id/analysis`?

mcous, 2022-04-18:
> > `GET /protocols/:id/analyses` returns full analyses
> 
> Should this be `/protocols/:id/analysis`?

"Analyses" as in the plural of "analysis". The protocol has a list field called `analyses`, which may contain 0 or more analysis objects

commit dfceb4b8 (2022-04-28): refactor(robot-server): move large analysis responses to own endpoint (#9992)

Closes #9965
closed 2022-04-28 by b-cooper