Opentrons/opentrons #9966: Delete old runs, protocols, and analyses

SyntaxColoring, 2022-04-14:
[@SyntaxColoring to flesh out details]

# Acceptance criteria

* A continually-used robot should not create persistent records at infinitum; old ones should be deleted.
* Consistency of the database and the API should be preserved. For example, we shouldn't leave runs that refer to protocols that no longer exist.

# See also

#9367

mcous, 2022-04-20:
I think this is a good spec. If we encounter any roadblocks during implementation, I believe we have an simplification offramp available that might help:

> If every existing protocol is linked to a run, the server can't safely delete any of them without violating the database's consistency. In this case, the server will repeatedly delete the oldest run that is linked to a protocol until any protocol is orphaned, and then delete that protocol.

We could potentially drop this requirement for simplicity and still maintain consistency, by treating the 20 protocol limit as soft and allowing the POST to succeed.

Then, one of two things will happen:

1. User will `POST /runs` using the new protocol
    - New run will cause the oldest run to be evicted, probably orphaning a protocol
    - Subsequent `POST /protocols` will evict that newly orphaned protocol
2. User will `POST /protocols` with another new protocol, without ever creating a run
    - Since the previously created protocol that exceeded the limit was never used in a run, it is deleted
    - New protocol is created

The simplification here might be worth it:

- `POST /runs` _never_ has to modify the protocols table
- `POST /protocols` _never_ has to modify the runs table

SyntaxColoring, 2022-04-20:
Interesting, yeah, that could work.

One possible downside is that, in a sense, it sort of degenerates to a maximum protocol limit of 1, I think.

1. We start with 20 protocols and 20 runs (mapped 1:1) uploaded.
2. Client A uploads protocol 21. This is over the soft limit, but doesn't evict anything because everything else is linked to a run.
3. Client B uploads protocol 22. This evicts protocol 21.
4. Client A tries to start a run with protocol 21, but it's already gone.

I *think* we could solve this in practice by making the protocol limit something like 40, keeping the run limit at 20? It's making my head hurt.

PR #10248 (2022-05-13): feat(robot-server): Automatically delete old resources to make room for new ones
# Overview

This PR closes #9966. See that issue for motivation and acceptance criteria.

# Changelog

Adopt the strategy for auto-deleting old runs and protocols that was described in #9966, [@mcous's reply to it](https://github.com/Opentrons/opentrons/issues/9966#issuecomment-1104285793), and [my reply to his reply](https://github.com/Opentrons/opentrons/issues/9966#issuecomment-1104332801).

Except for one thing. That specification says to auto-delete protocol *analyses* in addition to protocols and runs. This  [... cut]
commit 9ab070fa (2022-05-13): feat(robot-server): Automatically delete old resources to make room for new ones (#10248)

Closes #9966.
closed 2022-05-13 by SyntaxColoring