Opentrons/opentrons #9988: api: Thermocycler execute profile bug.

alexjoel42, 2022-04-18:
# Overview

We have an issue where there is a key error for any "execute_profile." 

# Steps to reproduce

1. Include tc_execute_profile with any quantity of steps on 5.x.x
2. Receive a software error 

```python3

metadata = {
    'protocolName': 'Thermocycler Example Protocol',
    'author': 'Opentrons <protocols@opentrons.com>',
    'source': 'Protocol Library',
    'apiLevel': '2.12'
    }

def run(protocol):
    tc_mod = protocol.load_module('thermocycler')
    # run profile
    profile = [
        {'temperature': 90, 'hold_time_seconds': 10}
    ]

    tc_mod.execute_profile(steps=profile, repetitions=1)

```

# Current behavior
```
Traceback (most recent call last):
  File "/Users/alexcopperman/opt/anaconda3/bin/opentrons_simulate", line 8, in <module>
    sys.exit(main())
  File "/Users/alexcopperman/opt/anaconda3/lib/python3.9/site-packages/opentrons/simulate.py", line 605, in main
    print(format_runlog(runlog))
  File "/Users/alexcopperman/opt/anaconda3/lib/python3.9/site-packages/opentrons/simulate.py", line 391, in format_runlog
    + command["payload"].get("text", "").format(**command["payload"])
KeyError: "'temperature'

```

# Expected behavior

Thermocycler starting 1 repetitions of cycle composed of the following steps: [{'temperature': 90, 'hold_time_seconds': 10}]

# work around 
install 4.7.0

amitlissack, 2022-04-18:
Same as #9799

closed 2022-04-21 by mcous

FankangMeng, 2022-04-23:
Same problem for me when simulation protocol with a thermocycler profile. Any solutions?🐶

Traceback (most recent call last):
  File "/Users/fankangmeng/opt/anaconda3/bin/opentrons_simulate", line 8, in <module>
    sys.exit(main())
  File "/Users/fankangmeng/opt/anaconda3/lib/python3.8/site-packages/opentrons/simulate.py", line 605, in main
    print(format_runlog(runlog))
  File "/Users/fankangmeng/opt/anaconda3/lib/python3.8/site-packages/opentrons/simulate.py", line 391, in format_runlog
    + command["payload"].get("text", "").format(**command["payload"])
KeyError: "'temperature'"

mcous, 2022-04-23:
@FankangMeng this will be resolved in the upcoming v5.1.0 release, which is currently undergoing testing

FankangMeng, 2022-04-23:
> @FankangMeng this will be resolved in the upcoming v5.1.0 release, which is currently undergoing testing

Thank you!

bndo, 2022-05-20:
Just a heads up @FankangMeng the next release will be `v6.0.0` (but this will still be part of that)