Opentrons/opentrons #9991: AP and PDI: Improve boundary condition warnings on pipetting flow rates

alexjoel42, 2022-04-18:
# Overview

Protocol Designer and Python API should clearly communicate pipettes' maximum and minimum flow rates. Right now the information is a bit disorganized. 

1. Python API doesn't give a maximum/minimum flow rate 
2. Our pipette[ GitHub documentation ](https://github.com/Opentrons/opentrons/blob/edge/shared-data/pipette/definitions/pipetteNameSpecs.json#L124)gives its own specification limitations (0.08 uL/sec to  24 uL/sec
3. Our Protocol Designer (0.1 uL/sec to 20 uL/sec) 
4. Testing internally 22.64088uL/sec

# Implementation details

- Establish true minimum and maximum for pipette flow rates @anuwrag 
- Sync maximum and minimum flow rates for Protocol Designer and Python API. 
- Decide on whether we want to sync default flow rates. Squashing [7652](https://github.com/Opentrons/opentrons/issues/7652)) is somewhat appealing, but could effect the reproducibility of Science if one day someone edits their PD protocol or changes it and the defaults all change. 
- During simulation, warn customers if they are going beyond the maximum flow rate speed (or below the minimum).

# Design
N/A 

# Acceptance criteria
Having all parts of our software agree on maximum and minimum flow rates.

ecormany, 2022-05-18:
Taking a closer look at this, not only do the API docs not state a minimum or maximum, the API itself does no validation other than making sure that the value is numeric and greater than zero. `left_pipette.flow_rate.aspirate = 99999` is completely accepted (and quite impossible).

The quick solution is to add the min and max values from the pipette specs file into documentation ([Jira ticket](https://opentrons.atlassian.net/browse/RTC-78)). The real fix is to get upper bound validation in the API itself.

alexjoel42, 2022-05-18:
Unfortunately, the quick solution doesn't solve all  of the underlying problems

1. PD and the pipette specs file have different maximum/minimum boundaries 
2. Our internal liquid handling team and pipette spec file  **do not match exactly** which is an embedded team 

The bug ticket's purpose was to ideally align embedded, liquid handling, Python API, Protocol Designer, and general documentation

I would definitely agree that a Jira ticket is an important step in the right direction though!

ecormany, 2022-05-18:
Agreed, adding to the API docs will not close this ticket.

jerader, 2024-06-03:
In ~February of 2023, we started having more robust pipette definitions (v2). While this data did not include max and min `flowRate`s, it did include default `flowRates` for certain API levels, and for `aspirate`, `dispense`, and `blowOut`. The API has been using this data but PD will start using this data in 8.1.

Earlier this year, having official ranges for `flowRate` in PD was scoped into the 8.1 PRD. That led me to doing a deep dive with Andy Sigler and Seth to see if we can make that happen. To summarize what we discussed, there is no risk of breaking a pipette if a `flowRate` is too low or too high but there is a point where the pipette can't physically reach a certain `flowRate`. So we wanted to define the max flow rate as the highest `flowRate` the pipette could physically reach. 

Due to the robustness of the v2 pipette definitions for GEN2 and Flex pipettes, the `flowRates` are based on the pipette version, the volume aspirated/dispensed/blown out, and the attached tip length. So for example with a p1000 using a 1000uL tip, there are 1000 different max `flowRates`. This became a bit overwhelming so we decided to keep it more simple where the max `flowRate` is a number that each volume can achieve based off of the equation. This is the `uiMaxFlowRate` used in PD (starting in 8.1) and Simulator (starting in 7.3.0) and the app (for some future features). (If you look at a pipette definition's liquid specs, you will find a `uiMaxFlowRate` key for each supported tip of a pipette. I think @ecormany added info about these ranges to the docs?). That being said, the GEN1 pipettes DO NOT have a `uiMaxFlowRate`. We kept those models out of the equation since they are older and their definitions don't have as much data as the GEN2 and Flex pipettes.

I think in the coming months we might further refine the `uiMaxFlowRate` equation, since we might have oversimplified it - I can keep you all informed if we do so. Please let me know if you have any questions!