Opentrons/opentrons #10015: 5.1 Feedback: Magnetic module Set engage height POST /commands missing param

y3rsh, 2022-04-19:
# Overview

On the set engage height slide out.

Current command shape is missing `moduleId: {id for the specific magnetic module returned from GET /modules}`

## Steps to reproduce

1. With a robot connected that has an attached magnetic module
2. Navigate to the Devices>Select connected robot
3. Use the kebab menu  on the magnetic module to "Set engage height"
4.  In the slide-out set the engage height to 10

## Current behavior

Engage height is not set, error in console.
POST /commands
```js
{
   "data":{
      "commandType":"magneticModule/engageMagnet",
      "params":{
         "engageHeight":10
      }
   }
}
```
## Expected behavior

Engage height is set.
POST /commands
```js
{
   "data":{
      "commandType":"magneticModule/engageMagnet",
      "params":{
         "moduleId":"MODULE_ID",
         "engageHeight":10
      }
   }
}
```

y3rsh, 2022-04-21:
I am seeing that the call to /modules is using `opentrons-version: 2` it will need to be `3` or `*` to get the module id to use in the command call.

sakibh, 2022-05-16:
Closing this ticket. Issue was addressed by #10119

closed 2022-05-16 by sakibh