Opentrons/opentrons #10021: bug: thermocycler module controls for lid and block both trigger the lid command when active

sakibh, 2022-04-20:
# Overview

When the lid and block of the thermocycler is active, clicking `deactivate lid` or `deactivate block` trigger the `thermocycler/deactivateLid` command type. The triggers need to be refactored so that the button onClicks are aware which command should be triggered.

[image]

# Steps to reproduce

- Run the robot-server and open the app in a dev environment. Then visit the `Devices` page to access the module controls.

mikecardarelli, 2022-05-17:
Block on receiving lid heating statuses from the robot server

mcous, 2022-05-18:
I don't think it's correct to say this issue is blocked by the lack of "heating"/"cooling" status for the lid. If the UI needs to know whether the lid heater is _enabled_ or _disabled_, it should look for the presence of the `data.lidTargetTemperature` in the modules response. If it is set, the lid is enabled and a `deactivateLid` command is appropriate. If the field is not present, the lid is deactivated and a `setTargetLidTemperature` is appropriate.

(The same is true for the block via `data.targetTemperature`)

PR #10350 (2022-05-19): refactor(app): fix thermocycler and H-S module controls

# Overview

This PR fixes the bug where clicking `deactivate block` would actually send a request to deactivate the lid. It also fixes a similar issue with the heater shaker controls where when the heater is active the menu would also think the shaker was active too. 

closes #10021

# Changelog

- Refactored `useModuleOverflowMenu`

# Review requests

- Clicking `deactivate lid` should send a deactivate lid command.
- Clicking `deactivate block` should send a deactivate block command.
- Setting a temperature on the heater shaker only sh [... cut]
commit 6dd0a2c7 (2022-05-19): refactor(app): fix thermocycler and H-S module controls (#10350)

This PR fixes the bug where deactivating the thermocycler block would deactivate the lid instead.

closes #10021
closed 2022-05-19 by sakibh