Opentrons/opentrons #10032: 5.1 Feedback:  Unavailable robots display robot settings and pipettes and modules

nusrat813, 2022-04-21:
### Current Behavior

If a robot that was previously online but is now offline, the robot will continue to display robot settings, pipettes and modules
[image]
[image]
 

### Expected Behavior

A user should not be able to see the contents of a robot if the robot is unavailable
[image]
 

### Operating system

Mac

nusrat813, 2022-04-29:
Robot server not being up - we want to make the settings still available

emilywools, 2022-05-11:
per figma, if a robot is offline, the user should not be able to access settings. 

https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=1297%3A91474

closed 2022-06-07 by b-cooper

y3rsh, 2022-06-08:
verified

commit 192b8e26 (2022-06-07): refactor(app): handle unreachable robots, time to load improvements on devices routes (#10399)

Do not show cached information about a robot that is unreachable on the devices landing, device
detail, or device settings page.

Closes #10032, Closes #10055