Opentrons/opentrons #10037: bug: Discrepancy in applied offsets when running protocol through SSH versus Jupyter

braldrich, 2022-04-21:
# Overview

Customer is applying labware offsets directly to protocol according to https://support.opentrons.com/en/articles/5901728-how-to-prepare-jupyter-notebook-and-opentrons_execute-for-5-0-0. 

They are reporting that the labware offsets are correctly applied in Jupyter but not with opentrons_execute on the command line. 

# Steps to reproduce

Note: I could not reproduce 

Follow https://support.opentrons.com/en/articles/5901728-how-to-prepare-jupyter-notebook-and-opentrons_execute-for-5-0-0 to download labware offsets and use set.offset to apply it in the protocol, then run in Jupyter and over SSH to see if the offsets are applied. 

# Current behavior

Customer's behavior: with labware offsets defined in the protocol, opentrons_execute ignores and does not apply the offsets. In Jupyter, it works 

# Expected behavior

It should work with Jupyter and opentrons_execute. 
[dispensecover.py.zip](https://github.com/Opentrons/opentrons/files/8534837/dispensecover.py.zip)

mcous, 2022-04-21:
Knowing the code in play, this problem as stated seems implausible. Pretty bad if true, so we'll investigate, but I really don't see how this could happen.

In the attached protocol, there are two different sets of offsets applied according to a `USING_SSH` flag. Is there a possibility this is the source of the user's observed differences?

```python
    if USING_SSH:
        # ...
        # Apply offests as needed
        tipRack.set_offset(x=0.00, y=9.00, z=0.10)
        suckerRack.set_offset(x=0.00, y=0.00, z=73.00)
        coverRack.set_offset(x=-1.00, y=14.20, z=-8.00)
        chipPlate.set_offset(x=-4.00, y=15.00, z=0.00)
        reservoir.set_offset(x=0.00, y=11.20, z=0.20) 
        
    else:
        # ...
        # Apply offsets
        tipRack.set_offset(x=0.00, y=0.40, z=0.00)
        suckerRack.set_offset(x=0.00, y=0.00, z=0.00)
        coverRack.set_offset(x=-1.00, y=1.00, z=-8.00)
        chipPlate.set_offset(x=-4.00, y=4.20, z=-0.80)
        reservoir.set_offset(x=0.10, y=0.70, z=1.10)
```

BSNXCody, 2022-04-22:
The offset under the “if” is there because those are the offset values that are needed to hit the target areas due to discrepancy.
The offset values under the “else” statement are from the labware position check.

BSNXCody, 2022-05-03:
Digging around I found the offset might be caused by not setting the environment variable RUNNING_ON_PI = 1.

My issue now is that applying os.environ['RUNNING_ON_PI'] = '1' in my protocol the robot still seems to think it is not changed
[image]
.

braldrich, 2022-05-16:
Hi @BSNXCody, hope you're doing well! If you are still seeing this issue, I would suggest reaching out to support@opentrons.com to get support for it. This seems like it's more an issue with the protocol than a software bug.

mcous, 2022-05-18:
Closing as unable to reproduce

closed 2022-05-18 by mcous