Opentrons/opentrons #10040: 5.1 Feedback:  Proceed to setup failing (POST /protocols form data malformed)

y3rsh, 2022-04-21:
### Overview

Proceed to setup through Devices fails.
Proceed to setup through Protocols fails.
On the network I see the POST to /protocols getting 500
> Hunch is that the finicky form-post data in the POST is not formatted correctly.  The integration tests that upload protocols are not having any issues. 
[image]

### Current Behavior

Nothing happens when you click "Proceed to setup"

### Expected Behavior

Proceed to setup takes us to setup.

### Steps To Reproduce

```gherkin
Given I have uploaded a simple protocol
And I have a robot connected on Wi-Fi
And I navigate to Devices
And I expand my connected robot
And I Click "Run a protocol"
And on the slide out I choose the simple protocol (selected by default when I have just one uploaded protocol)
When I click "Proceed to setup"
Then nothing happens
```

### Operating system

Mac

### Robot setup or anything else?

[protocol](https://github.com/Opentrons/opentrons/files/8535329/simple.json.zip)
error returned
```js
{
   "errors":[
      {
         "id":"UnexpectedError",
         "title":"Unexpected Internal Error",
         "detail":"opentrons.protocol_reader.protocol_reader.ProtocolFilesInvalidError: No files were provided.",
         "meta":{
            "stacktrace":"Traceback (most recent call last):\n  File \"usr/lib/python3.7/site-packages/robot_server/protocols/protocol_store.py\", line 325, in _compute_protocol_sources\n  File \"usr/lib/python3.7/site-packages/anyio/_backends/_asyncio.py\", line 564, in __aexit__\n  File \"usr/lib/python3.7/site-packages/anyio/_backends/_asyncio.py\", line 601, in _run_wrapped_task\n  File \"usr/lib/python3.7/site-packages/robot_server/protocols/protocol_store.py\", line 315, in compute_source\n  File \"usr/lib/python3.7/site-packages/opentrons/protocol_reader/protocol_reader.py\", line 105, in read_saved\nopentrons.protocol_reader.protocol_reader.ProtocolFilesInvalidError: No files were provided."
         }
      }
   ]
}
```

emilywools, 2022-04-26:
yes we should provide feedback to the user that this button was clicked. we could show a loader inside the button while this is happening. @mmencarelli do you have any thoughts here?

mmencarelli, 2022-04-29:
@emilywools When the user clicks 'proceed to setup' the app performs an analysis of the protocol which takes a few seconds, which is what I think @y3rsh is encountering there. 

We have a loading state for this as a solution (screenshot below), but I don't think it's been implemented yet. Right, @b-cooper ? 

[image]

nusrat813, 2022-05-11:
@b-cooper - can you look at the design provided by mel?

y3rsh, 2022-05-26:
Now that we are progressing to the run setup page and have the analyzing on robot spinner, this experience is much improved.

closed 2022-05-26 by y3rsh