Opentrons/opentrons #10041: 5.1 Feedback: Robot update warning remains after update

ecormany, 2022-04-21:
### Overview

The robot server update warning in the robot card on the devices tab does not dismiss itself after successfully completing the update flow, nor does the success toast message appear. The warning can be dismissed manually and will not reappear after a successful update.

### Current Behavior

Upon completing the robot server update, nothing changes in the devices view, including the presence of the update warning. Clicking on "View update" informs you that the robot is in fact up to date.

https://user-images.githubusercontent.com/1958332/164560578-ad94d461-4537-4443-af86-16027632142b.mov

(Apologies that the completed update modal only appears for the first couple frames of this screen recording.)

### Expected Behavior

The warning should disappear and the success toast should appear temporarily.

### Steps To Reproduce

1. Open the 5.1.0 beta app.
2. Connect a robot with a server version less than v5.1.0-beta.0.
3. Click "View Update" on the robot card and follow the update steps.
4. Dismiss the modal that says "Your robot is now successfully updated."

I reproduced this twice by downgrading the robot to 5.0.2 and relaunching the 5.1.0 beta app.

### Operating system

Mac

### Robot setup or anything else?

Update was performed over Wi-Fi.

mcous, 2022-05-24:
Still present on `edge` as of a731395

[image]

nusrat813, 2022-05-26:
Still present on 6.0.0-alpha.0

emilywools, 2022-06-01:
@bndo i think @jerader is working on the overflow menu item right now on another ticket

commit 1d4a1528 (2022-06-01): fix(app): fix robot update banner

closes #10041
PR #10567 (2022-06-01): fix(app): fix robot update banner
closes #10041

# Overview

this PR fixes the bug where the robot update banner persists even after the robot has been updated. 

# Changelog

- edit logic in `UpdateRobotBanner` and create test

# Review requests

- if your robot needs to be updated, update it. After its complete, the banner should disappear. 

# Risk assessment

low
commit a7713c45 (2022-06-01): fix(app): fix robot update banner (#10567)

closes #10041
closed 2022-06-01 by jerader