Opentrons/opentrons #10047: 5.1 Feedback:  A single protocol can be uploaded many times

nusrat813, 2022-04-22:
### Current Behavior

A single protocol can be uploaded many times  - the app does not reanalyze the file as one file 
[image]

### Expected Behavior

Is the expected behavior supposed to be like the custom labware upload where there is a modal that says do you want to overwrite the file? 

### Steps To Reproduce

1. Upload a protocol 
2. Upload the same protocol multiple times

### Operating system

Mac

nusrat813, 2022-04-28:
See expected behavior comment 
@emilywools - do you have any insight on this?

nusrat813, 2022-04-29:
Expected behavior at the moment 
Feature request - not in scope for 5.1 currently @mikecardarelli -to ticket

closed 2022-04-29 by nusrat813

mikecardarelli, 2022-05-02:
New ticket https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/10181