Opentrons/opentrons #10054: 5.1 Feedback: many UI elements offer "Copy Link" contextual menu item

ecormany, 2022-04-22:
### Overview

Most UI elements in the new app UI can spawn a contextual menu item with a "Copy Link" item.

[image]
[image]
[image]
[image]

### Current Behavior

Most UI elements in the new app UI can spawn a contextual menu item, which usually only contains a "Copy Link" item (and also "Copy" if the text is selectable). This item does in fact copy a `file:///` URL to the clipboard, but its contents serve no purpose to the end user.

### Expected Behavior

These contextual menus should not appear if their only item is "Copy Link". Alternatively, if the URLs produced by Copy Link serve some internal or debugging purpose, the feature should be enabled by advanced preference / feature flag only.

### Steps To Reproduce

1. Launch the Opentrons App.
2. Right click on certain interface elements. These include (but are not limited to) sidebar text items, sidebar icons, protocol cards, and robot cards. Labware cards do _not_ spawn these menus.
3. Click Copy Link.

A link similar to `file:///Applications/Opentrons.app/Contents/Resources/app.asar/ui/index.html#/robots` will be put on the clipboard. This URL is meaningless because the app.asar file is not a bundle whose contents can be viewed individually.

Note: This behavior is also present in earlier versions of the app; it just manifests on significantly fewer UI elements.

### Operating system

Mac

emilywools, 2022-05-02:
we would like to remove all copy link context menus

koji, 2022-05-16:
@ecormany 
Do you think we should remove look up too?

[image]

ecormany, 2022-05-17:
"Look Up" is a standard OS option and most importantly does what it claims to do, so I think we can leave it. "Copy Link" doesn't copy a functioning link to anything.

closed 2022-05-24 by koji

y3rsh, 2022-05-26:
On mac[image]
On Windoes 11 and Linux I get only Inspect element.

PR #10328 (2022-05-24): fix(app): remove copy link from context menu

# Overview
This will fix #10054
Removed `copy link` by using the `menu` function
https://github.com/sindresorhus/electron-context-menu#menu.

There is a kind of issue with `electron-context-menu` in this project.

The version app-shell uses is 0.15.0 (released in Aug 2019). The current version is 3.1.2.
For safe, I would like to upgrade this package by another PR.
 
In addition, 0.15.0 has an issue with `searchWithGoogle` option.

[image]

[image]

# Changelog
- Updated `contextMenu` option