Opentrons/opentrons #10055: 5.1 Feedback:  app goes white screen when clicked on Devices tab, after uploading a protocol

reachojhaneha, 2022-04-25:
### Overview

When a user clicks on device tab after uploading a protocol -> app goes balnk

### Current Behavior

App is blank when the user clicks on Devices tab - beta 5.1.0-beta.0
[image]

[image]

[image]

### Expected Behavior

App should not go blank

### Steps To Reproduce

1. Goto protocol tab and upload a protocol
2. Before the protocol is fully analyzed and the deckmap is displayed on protocol page -> user clicks on device page
3.  The app goes balnk. I am able to see the analysis.json on my mac `Library/Application Support/Opentrons/protocols/123fd6ca-3bd6-4d92-afc4-195e9ebb411f/analysis`

nusrat813, 2022-04-28:
@hartm9 - Can you ticket your comment as a separate issue?

nusrat813, 2022-04-28:
Do we need to make a decision about disabling tabs during certain analyses? (ex: protocol analyzing, proceed to setup, rerun)

b-cooper, 2022-05-11:
We do not need to disable tabs during analyses.  This a bug that has to do with the "Choose a Protocol Slideout" (which is part of the Devices tab) is not properly handling in progress analyses.

y3rsh, 2022-05-24:
I can still reproduce this
[image]

mcous, 2022-06-01:
This ticket interacts with #10468. If you quit out of the app while the protocol is still spinning, the protocol will never get an analysis object, which will cause this same whitescreen

closed 2022-06-07 by b-cooper

y3rsh, 2022-06-07:
Verified fixed.

commit 192b8e26 (2022-06-07): refactor(app): handle unreachable robots, time to load improvements on devices routes (#10399)

Do not show cached information about a robot that is unreachable on the devices landing, device
detail, or device settings page.

Closes #10032, Closes #10055