Opentrons/opentrons #10056: 5.1 Feedback:  Unable to clear protocol after run is cancelled - 409 conflict error appears

reachojhaneha, 2022-04-25:
### Overview

Cancel a running protocol and then clear the protocol. 409 conflict error appears

### Current Behavior

[image]
[image]

### Expected Behavior

The protocol should be cleared for the protocol to be run again.

### Steps To Reproduce

1. Upload a protocol from protocols tab
2. Goto device tab, select the device -> protocol -> run the protocol from device detail page.
3.  Cancel the run 
4. Now click on **clear protocol** button -> unable to clear protocol. 409 conflict error appears.
[image]

nusrat813, 2022-04-29:
@mcous - to investigate

mcous, 2022-05-03:
@reachojhaneha we've been unable to reproduce this issue, and we need the following information to investigate any further:

- The protocol used in the screen recording
- The response body of the 409 response

Additionally, the reproduction steps say the protocol should be cancelled, but the screen recording shows a protocol with the status "completed". Do you have any idea why this is the case?

reachojhaneha, 2022-05-03:
@mcous , here is the protocol used
[script_pur_sample_1.json.zip](https://github.com/Opentrons/opentrons/files/8613806/script_pur_sample_1.json.zip)

The response body of the 409 : 
{"errors":[{"id":"RunActionNotAllowed","title":"Run Action Not Allowed","detail":"Cannot modify a stopped engine."}]}

[image]

Screen recording has a completed protocol, the heading of the protocol says this issue occurs when protocol is completed or cancelled. I can modify the steps, i saw this issue with cancelled protocol first and then with completed protocols also.

mcous, 2022-05-04:
@reachojhaneha that new screenshot is a different request that originally specified in the issue. The original issue notes the `PATCH /runs/:run_id`, which is triggered by the "Clear protocol" button. The new screenshot is `POST /runs/:run_id/actions`, which is triggered by the "pause" and "cancel" buttons.

This second 409 is the server correctly rejecting a request to cancel a run after it's already completed, probably due to the protocol finishing naturally just before the cancel button was clicked / cancel request was received by the robot. From an app standpoint, it looks like the client handled this rejection properly, too, and is displaying the run as completed. The "error" in the devtools is just because Chrome always logs non-2xx responses in the devtools, even if the application handled it properly.

After trying for several days, we've been unable to reproduce server rejections of the "clear protocol" button request (`PATCH /runs/:run_id`) on `edge`, using both real hardware and the simulating server. Given that this UI is apparently changing once more, I'm going to close this issue in favor of opening a new one if we encounter buggy behavior with the new UI.

closed 2022-05-04 by mcous