Opentrons/opentrons #10058: 5.1 Feedback:  protocol that client-side py can't analyze

emilywools, 2022-04-25:
### Overview

[Test_Speaker.py.zip](https://github.com/Opentrons/opentrons/files/8555781/Test_Speaker.py.zip)

the app won't analyze this protocol, not sure why. its in a forever-loading state.

[image]
 

### Current Behavior

when i import this protocol into the app, the card appears but the protocol never fully analyzes.

### Expected Behavior

i expect this protocol to analyze and show deck state info and i can click into the detail page

### Steps To Reproduce

import the attached protocol to the beta

### Operating system

Mac

nusrat813, 2022-04-28:
@y3rsh can you download and try this on your robot?

y3rsh, 2022-04-29:
> @y3rsh can you download and try this on your robot?

This protocol references the a full robot file system path directly `/etc/audio/speaker-test.mp3` which would be a problem but the way it is referenced in this case does **NOT** cause the problem.
A bad path is masked with this code, no exception is raised.
```python
AUDIO_FILE_PATH = '/etc/audio/speaker-test.mp3'
def run_quiet_process(command):
    subprocess.check_output('{} &> /dev/null'.format(command), shell=True)
def test_speaker():
    print('Speaker')
    print('Next\t--> CTRL-C')
    try:
        run_quiet_process('mpg123 {}'.format(AUDIO_FILE_PATH))
    except KeyboardInterrupt:
        pass
        print()
```

mcous, 2022-05-03:
Moved to #10142

closed 2022-05-03 by mcous