Opentrons/opentrons #10062: 5.1 Feedback: TITLE UNSET

y3rsh, 2022-04-25:
# Feedback 
1. Not able to rename my robot. When I click in the text editor to type a name, it exits from that field. 
2. My robot serial number is not listed under “Robot Serial Number” instead it says “nameless-cloud”. 
4. When I load a protocol and click into that protocol, I have 2 tabs. Robot configuration and labware. The labware tab doesn’t contain anything. 
5. If I am only connected to one robot, why do I select which robot I am going to run on? Also, once I click on a robot, there is a long wait time, where it is not apparent that anything is happening, so I clicked that button multiple times.
6. Not a fan of the button layout. It’s not clear that I have the option to run the labware calibration prior to starting the run - it looks like I should just start the run. 
7. If you cancel a run, the tips get dropped. It would be good to have an option to empty the tips at a specific location before the drop. 
8. “Clear Protocol” functionality is really confusing. Clear Protocol of what exactly?
9. Stopped a protocol and the offsets were saved, but my partial run was not stored in the “recent protocol runs”.  under the devices tabs. 
10. Also under devices, the “Run a Protocol” button is greyed out. Why?
11. I uploaded multiple protocols. In one instance, I uploaded one protocol with the name Johns Hopkins, and another Johns Hopkins (1), but in the app, they populate with the same name, so I can’t tell the difference. 
12. I ran a protocol and ran an LPC with a protocol titled Johns Hopkins. I added an additional transfer step, but did not change the labware and reloaded the protocol using the same name. The labware offsets were not saved.
13. If I rerun multiple protocols of the same name they load in the app in a confusing order. I show a lot of duplicates, but I have to keep track of the timestamps. That is not ideal. 

## Files 
https://drive.google.com/open?id=13eZ1PMb17DXp6VgxZV4fq8gdXApZOBb6, https://drive.google.com/open?id=1sYJKFH4BrYI_g-NKzVov0yEJcZZebeV3, https://drive.google.com/open?id=1E-7d-qhb9fONMMTc9qcySYq8CZjxN6bU, https://drive.google.com/open?id=1L-7X6sW4NoGPj7-bCV7U9iQ_ttBaYgWv, https://drive.google.com/open?id=1YnOIFFcUPCeR7WJkxdau1tAAlFy9-TP-

nusrat813, 2022-04-28:
@y3rsh - task: break down feedback into multiple tickets as needed

y3rsh, 2022-04-29:
### Most aggregated to https://github.com/Opentrons/opentrons/issues/10130

> # Feedback
> 1. ~~Not able to rename my robot. When I click in the text editor to type a name, it exits from that field.~~ duplicate
> 2. My robot serial number is not listed under “Robot Serial Number” instead it says “nameless-cloud”.
> 3. ~~When I load a protocol and click into that protocol, I have 2 tabs. Robot configuration and labware. The labware tab doesn’t contain anything.~~  in dev
> 4. If I am only connected to one robot, why do I select which robot I am going to run on? ~~Also, once I click on a robot, there is a long wait time, where it is not apparent that anything is happening, so I clicked that button multiple times.~~ duplicate
> 5. Not a fan of the button layout. It’s not clear that I have the option to run the labware calibration prior to starting the run - it looks like I should just start the run.
> 6. If you cancel a run, the tips get dropped. It would be good to have an option to empty the tips at a specific location before the drop.
> 7. “Clear Protocol” functionality is really confusing. Clear Protocol of what exactly?
> 8. ~~Stopped a protocol and the offsets were saved, but my partial run was not stored in the “recent protocol runs”.  under the devices tabs.~~ known in dev
> 9. Also under devices, the “Run a Protocol” button is greyed out. Why?
> 10. I uploaded multiple protocols. In one instance, I uploaded one protocol with the name Johns Hopkins, and another Johns Hopkins (1), but in the app, they populate with the same name, so I can’t tell the difference. https://github.com/Opentrons/opentrons/issues/10129
> 11. ~~I ran a protocol and ran an LPC with a protocol titled Johns Hopkins. I added an additional transfer step, but did not change the labware and reloaded the protocol using the same name. The labware offsets were not saved.~~ LPC enhancement in dev
> 12. If I rerun multiple protocols of the same name they load in the app in a confusing order. I show a lot of duplicates, but I have to keep track of the timestamps. That is not ideal. https://github.com/Opentrons/opentrons/issues/10131
>