Opentrons/opentrons #10064: 5.1 Feedback:  Protocol Analysis Failure: libcrypt.so.1: cannot open shared object file: No such file or directory

ethanfjones, 2022-04-25:
### Overview

When I try to upload a simple protocol

from opentrons import protocol_api

metadata = {'apiLevel': '2.12'}

```
def run(protocol: protocol_api.ProtocolContext):
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', 1)
    tiprack_1 = protocol.load_labware('opentrons_96_tiprack_300ul', 2)
    p300 = protocol.load_instrument('p300_single_gen2', 'right', tip_racks=[tiprack_1])

    p300.pick_up_tip()
    p300.aspirate(100, plate['A1'])
    p300.dispense(100, plate['B1'])
    p300.drop_tip()
```

the below error is throw

> Command failed with exit code 127: /tmp/.mount_OpentrFxLjs0/resources/python/bin/python3 -m opentrons.cli analyze --json
> /home/opentrons/.config/Opentrons/protocols/cbc58829-9713-417b-b60c-dedacce2d0a6/src
> /tmp/.mount_OpentrFxljs0/resources/python/bin/python3: error while loading shared libraries: libcrypt.so.1: cannot open shared object file: No such file or directory

[image]

I am running the app on Manjaro 21.2.6

### Current Behavior

A protocol analysis error is thrown when I upload a protocol

### Expected Behavior

The protocol should upload without an error being thrown 

### Steps To Reproduce

Try to upload any protocol 

### Operating system

Linux

### Robot setup or anything else?

Wi-Fi 

I was able to resolve this by installing libxcrypt via AUR

nusrat813, 2022-04-28:
@nusrat813 - task: try to repo

nusrat813, 2022-04-29:
Need for support documentation for python packages 

#9788 - app setting to overwrite path to python

nusrat813, 2022-05-18:
I was not able to repo this issue

b-cooper, 2022-05-18:
Closing as this appears to be fixed

closed 2022-05-18 by b-cooper