Opentrons/opentrons #10070: Add 3 new PRL tube rack definitions to the Labware Library

skyewindsor, 2022-04-26:
# Overview

We need to add the following PRL Tube rack definitions to the Labware Library: 

1. Opentrons 32 Tube Rack with NEST 15 mL Conical ([labware definition](https://drive.google.com/drive/u/1/folders/1N8ICG36mOfpd2tLujCofXYRb7rGnVZme) / [image of rack](https://cdn11.bigcommerce.com/s-zmwgbev7rb/images/stencil/1280x1280/products/201/541/BLUE_31x15_36mm_spaced_Tube_Rack__64602.1647288508.png?c=1) / [image of tube](https://cdn11.bigcommerce.com/s-zmwgbev7rb/images/stencil/1280x1280/products/150/468/NEST_15mL_Centrifuge_Tube_1__96199.1638770249.png?c=1) / [rack website link](https://shop.opentrons.com/tube-rack-31-x-15m-36mm-spaced/) / [tubes website link](https://www.cell-nest.com/page94?_l=en&product_id=110) / Rack catalog number = 417-00081  / Tube catalog numbers = 601052)
2. Opentrons 32 Tube Rack with NEST 10 mL Vials ([labware definition](https://drive.google.com/drive/u/1/folders/109Q3g-z_HP2ywsqoBQB1wR6j98oUbBjn) / [image of rack](https://cdn11.bigcommerce.com/s-zmwgbev7rb/images/stencil/1280x1280/products/201/541/BLUE_31x15_36mm_spaced_Tube_Rack__64602.1647288508.png?c=1) / [image of tube](https://cdn11.bigcommerce.com/s-w9bdixgj/images/stencil/1280x1280/products/3503/7847/NEST_Scientific_202117_10mL_Vial_with_3mL_VTM_for_Storing_Patient_Samples_for_RT-PCR_Testing_of_Coronavirus_-_COVID-19_Testing_Supplies_-_Stellar_Scientific__58465.1596469583.jpg?c=2) / [rack website link](https://shop.opentrons.com/tube-rack-31-x-15m-36mm-spaced/) / [tube website link](https://www.cell-nest.com/page94?_l=en&product_id=754) / Rack catalog number = 417-00081  / Tube catalog numbers = NST-202017, NST-202117)
3. Opentrons 32 Tube Rack with Falcon 15 mL Conical ([labware definition](https://drive.google.com/drive/u/1/folders/1tGg0H3nySFLa7WjxhqN-qPfScndnnZ6-) / [image of rack](https://cdn11.bigcommerce.com/s-zmwgbev7rb/images/stencil/1280x1280/products/201/541/BLUE_31x15_36mm_spaced_Tube_Rack__64602.1647288508.png?c=1) / [image of tube](https://labware.opentrons.com/images/falcon_15ml_conical_tube.954ce93044fba21528d60596ad00ceb7.jpg) / [rack website link](https://shop.opentrons.com/tube-rack-31-x-15m-36mm-spaced/) / [tube website link](https://ecatalog.corning.com/life-sciences/b2c/US/en/Liquid-Handling/Tubes,-Liquid-Handling/Centrifuge-Tubes/Falcon%C2%AE-Conical-Centrifuge-Tubes/p/falconConicalTubes) / Rack catalog number = 417-00081 / Tube catalog numbers = 352095, 352096, 352097, 352099, 352196)

# Acceptance criteria

The 3 new PRL Tube Rack definitions appear in the labware library and render correctly (with display name, definition data, website link, and image):
- [ ] Opentrons 32 Tube Rack with NEST 15 mL Conical
- [ ] Opentrons 32 Tube Rack with NEST 10 mL Vials
- [ ] Opentrons 32 Tube Rack with Falcon 15 mL Conical

mikecardarelli, 2022-08-16:
Note: A new ticket for this has been added to JIRA (RAUT-146)

mikecardarelli, 2022-09-09:
Updated link for the new JIRA ticket: https://opentrons.atlassian.net/browse/RLAB-103

closed 2022-09-09 by mikecardarelli