Opentrons/opentrons #10080: 5.1 Feedback: Looking for robots loader

emilywools, 2022-04-26:
### Overview

on app launch, the devices tab of the app shows a "looking for robots" loader. Can we get rid of this? Is there a reason we need to show this? If we need to show it, can we restyle it to be smaller?
[image]

### Current Behavior

on app launch, the devices tab of the app shows a "looking for robots" loader.

### Expected Behavior

i dont think we need to show this loaded on app launch

### Steps To Reproduce

open the app

### Operating system

Mac

nusrat813, 2022-04-29:
No reason to show

closed 2022-05-17 by y3rsh

nusrat813, 2022-06-08:
Verified fixed

PR #10165 (2022-05-03): refactor(app): Remove loader from Devices page

# Overview
Remove from Looking for robots... loader from Devices page. 
This fixes #10080

# Changelog
- Removed the Scanning component from DeviceLanding component
- Removed a test case for Scanning component 

# Review requests
- No loader when open Devices page

# Risk assessment
low
commit 20eb5d5f (2022-05-03): refactor(app): Remove loader from Devices page (#10165)

* refactor: romove loader from Devices page

Remove from Looking for robots... loader from Devices page. This fixes #10080

##10080

* Removed Scanning component