Opentrons/opentrons #10085: 5.1 Feedback:  See how to setup a new robot modal

emilywools, 2022-04-26:
### Overview

the link to display the modal, as well as the modal styling is a bit off

[image]
[image]

### Current Behavior

link:
the "see how to set up a new robot" link font size is too big

modal:
Close is not capitalized
Not enough space between the body copy and the link to the help article

### Expected Behavior

link:
should be 11pt

modal:
should capitalize Close
needs 16px between body copy and link

### Steps To Reproduce

go to devices page
click "See how to set up a new robot"

### Operating system

Mac

koji, 2022-04-26:
- [ ] Add css(TYPOGRAPHY) to Link
- [ ] Capitalize `close`
- [ ] Update Modal

closed 2022-05-17 by y3rsh

y3rsh, 2022-05-26:
[image]

[image]

PR #10090 (2022-04-28): fix(app): Updated See how to setup a new robot modal styling

# Overview
Fixes #10085 

[image]

[image]

# Changelog
- Add css(TYPOGRAPHY) to Link
- Capitalize modal close button's first letter
- Add stying to Modal child

# Review requests

- link: font size 11pt
- modal: capitalize close
-