Opentrons/opentrons #10086: 5.1 Feedback:  Protocols > Protocol Detail overflow menu persists after no selection

y3rsh, 2022-04-26:
### Overview

When I open the overflow menu on a protocol detail it does not disappear when I click off of it.  The only way it collapses without selecting an item in the menu is to click the overflow menu kebab again.

### Current Behavior

https://user-images.githubusercontent.com/502770/165359299-cf1e30aa-8570-4042-9a47-a9c9ff5c0fb5.mp4

### Expected Behavior

Clicking off of the the menu or clicking the kebab would close the menu.

### Steps To Reproduce

See the video.

### Operating system

Mac

nusrat813, 2022-04-28:
Overflow menus in the devices tab disappear when you click off of them - we should have the same behavior for the protocol detail overflow menu

closed 2022-05-19 by sakibh

y3rsh, 2022-05-26:
verified
[image]

PR #10296 (2022-05-19): refactor(app): add hook for overflow menu outside click
# Overview

This PR abstracts the logic into a hook in order to hide the overflow menu when an outside node is clicked.

closes #10086, closes #10209

# Changelog

- Added `useMenuHandleClickOutside`

# Review requests

- Check that when an overflow menu is clicked and then clicking outside of the menu closes it.

# Risk assessment

low
commit 096786eb (2022-05-19): refactor(app): add hook for overflow menu outside click (#10296)

* refactor(app): add hook for overflow menu outside click

This PR abstracts the logic to hide the overflow menu when an outside node is clicked.

closes #10086