Opentrons/opentrons #10087: 5.1 Feedback:  Long Protocol Name cuts off the X button on "Run protocol" slideout

y3rsh, 2022-04-26:
### Overview

With a sufficiently long protocol name like `MagneticV2TempV2MixTransferNoPause`
The X button on the "Run protocol" slideout gets cut off
[image]

### Current Behavior

X cut off

### Expected Behavior

X not cut off

### Steps To Reproduce

Import a protocol with a long name.
[MagneticV2TempV2MixTransferNoPause.json.zip](https://github.com/Opentrons/opentrons/files/8565630/MagneticV2TempV2MixTransferNoPause.json.zip)
 Go to protocol detail  and click run a protoco

https://user-images.githubusercontent.com/502770/165360482-885819b0-f991-43e9-9813-5690c44088d8.mp4

l

### Operating system

Mac

emilywools, 2022-05-02:
@mmencarelli can you weigh in here?

mmencarelli, 2022-05-05:
There should be a 8px margin between the title and the 'x' button. I think this is happening because there's no space in the name of the robot. Could we force a break in the name somehow?

nusrat813, 2022-05-11:
@mmencarelli - was a final decision made on this?

emilywools, 2022-05-11:
@nusrat813 i think we need someone from SWE to weigh in on this.

koji, 2022-05-20:
[image]

closed 2022-05-24 by koji

y3rsh, 2022-05-26:
verified
[image]