Opentrons/opentrons #10091: 5.1 Feedback:  Cursor doesn't change on labware cards

emilywools, 2022-04-26:
### Overview

when you hover over a labware card on the labware tab, the cursor does not change to a finger (is that the right word????)

### Current Behavior

when you hover over a labware card on the labware tab, the cursor does not change to a finger (is that the right word????)

### Expected Behavior

when you hover over a labware card on the labware tab, the cursor should change to a finger

### Steps To Reproduce

go to labware page and hover over a labware card

### Operating system

Mac

koji, 2022-04-26:
@emilywools 
finger = `pointer` in css
https://developer.mozilla.org/en-US/docs/Web/CSS/cursor

closed 2022-05-17 by y3rsh

nusrat813, 2022-06-08:
verified fixed

PR #10101 (2022-04-28): fix(app): Modify labware card and slideout style

# Overview
This fixed #10091, #10092, #10093

# Changelog
- Added `cursor` to LAYOUT_PROPS(style-props.ts)
- Added a new icon, `copy-text`
- Changed the icon from `ot-copy-text` to `copy-text`
- Prevent opening the detail slideout when clicking `copy-text` icon

# Review requests

# Risk assessment
low