Opentrons/opentrons #10092: 5.1 Feedback:  copy button on API name on labware card does not seem clickable

emilywools, 2022-04-26:
### Overview

copy button on API name on labware card is not clickable

### Current Behavior

if you click the copy icon on the labware card, it copies the API name to the clipboard, but you would not know it because it opens the labware details slideout. 

### Expected Behavior

When you click the copy icon on the labware card, it should give the user visual feedback that the API name was copied. this action should not open the labware details slideout

### Steps To Reproduce

go to labware page. click the copy icon on a labware card

### Operating system

Mac

koji, 2022-04-26:
- [ ] update copy icon (copy icon is different from the design)
  - [ ] add a new icon to icon-data.ts

koji, 2022-04-29:
This has been solved by #10101

closed 2022-04-29 by koji