Opentrons/opentrons #10093: 5.1 Feedback:  labware card: copy API name icon does not match Figma

emilywools, 2022-04-26:
### Overview

labware card: copy API name icon does not match Figma

### Current Behavior

the copy icon on the labware card shows a clipboard icon.

### Expected Behavior

The copy icon on the labware card should show the new copy icon

### Steps To Reproduce

go to labware page and look at the copy icon on the labware card

### Operating system

Mac

emilywools, 2022-04-26:
link to figma node: https://www.figma.com/file/l7BAJAGICr10ELsSqQD0Sr/UI-Overhaul?node-id=2180%3A85399

koji, 2022-04-29:
This has been fixed by #10101

closed 2022-04-29 by koji

y3rsh, 2022-05-26:
[image]