Opentrons/opentrons #10094: 5.1 Feedback: mounts described as "empty" when they aren't or don't need to be

ecormany, 2022-04-26:
### Overview

The app is too liberal in using the term "Empty" for pipette mounts. We should adopt clearer language on the Devices and Protocols page.

### Current Behavior

### Robots page
On the robots page, both mounts for unavailable robots are listed as "Empty". Since the robots aren't in communication with the app, that information may be incorrect (and likely is incorrect in settings where users don't remove their pipettes).
[image]

### Protocols page
If a protocol only uses a single pipette, it lists the other mount as "empty". This makes it seem like the protocol _requires_ that the other pipette be removed in order to run it, when this is not the case.
[image]

### Expected Behavior

### Robots page
The mount sections should _either_ show:
1. The pipettes that were in place when the robot was last seen.
2. Some text expressing that the information is not available: e.g. "Unknown" or an em dash

### Protocol page
The unused mount should say "Unused" or "Not used".

### Steps To Reproduce

View any unavailable robot on the Robots page or any loaded protocol on the Protocols page.

### Operating system

Mac

koji, 2022-04-28:
@emilywools @mmencarelli 
What do you think about this?

Another question `empty` 's first letter should be a capital letter?

nusrat813, 2022-04-28:
Addressing Ed's expected behavior comment #1 - https://github.com/Opentrons/opentrons/issues/10032

nusrat813, 2022-04-29:
@mikecardarelli - to ticket 
protocol page - unused 
wording for empty not needed because section will be taken out

mikecardarelli, 2022-05-02:
New ticket for "Not used" pipettes: https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/10180

nusrat813, 2022-05-11:
PR Merged - https://app.zenhub.com/workspaces/app--ui-5f6248dba845b000139aa767/issues/opentrons/opentrons/10188

y3rsh, 2022-06-08:
I think this is good
[image]
[image]