Opentrons/opentrons #10095: 5.1 Feedback: radio button component does not match figma

emilywools, 2022-04-26:
### Overview

the radio button on the App Settings > Advanced > Tip length calibration settings do not match figma
[image]

### Current Behavior

the radio buttons on the tip length calibration settings do not match the figma designs, the selected radio button is gray

### Expected Behavior

the selected radio button should be blue with a blue outline. see this figma node: https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=1297%3A88764

### Steps To Reproduce

go to app settings > advanced

### Operating system

Mac

koji, 2022-04-27:
This will be solved by #9767

[image]

closed 2022-05-11 by nusrat813

y3rsh, 2022-06-08:
verified in alpha 1
[image]