Opentrons/opentrons #10098: 5.1 Feedback:  Run option is available for a protocol with failed analysis

reachojhaneha, 2022-04-26:
### Overview

Click on the overflow menu of the protocol with failed analysis -> Run option is available.

### Current Behavior

[image]

### Expected Behavior

[image]

### Steps To Reproduce

1. Click on the overflow menu of a protocol with failed analysis
2. Run option is available
3. click on the run button - and choose the robot to run the protocol on -click on proceed to setup
4. Robot is busy pop-up appears - clicking on clear and proceed to setup does nothing.
5. Expected behavior is the run option should be disabled

emilywools, 2022-04-27:
https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=8250%3A330560

reachojhaneha, 2022-04-27:
Thanks @emilywools , if run option should be available for the protocol with failed analysis, i am getting a 422 error on POST request on /protocols api. 
[image]

At the UI level, there is no response. App is not responding at all.

I will change the ticket with this 422 error issue

nusrat813, 2022-04-29:
@reachojhaneha - please attach the file used for this error 
@mcous - try to repo 
@mikecardarelli - ticket for run creation failure (protocol + run creation bundle)

reachojhaneha, 2022-04-29:
[upload_protocol.json.zip](https://github.com/Opentrons/opentrons/files/8594136/upload_protocol.json.zip)

mikecardarelli, 2022-05-02:
@nusrat813 Do you remember what exactly needed a new ticket here? I can't remember what we decided on Friday.

closed 2022-05-11 by nusrat813