Opentrons/opentrons #10099: 5.1 Feedback:  <Robot> is busy modal appears twice after clicking on clear and proceed to setup button

reachojhaneha, 2022-04-26:
### Overview

If the user chooses a robot that has a protocol loaded and in the not started, complete, canceled or failed state,  **robot is busy**  modal appears

### Current Behavior

this modal appears twice, second time it vanishes in seconds
[image]

after clicking on clear and proceed to setup, again a modal appears with robot is busy and disappears
[image]

### Expected Behavior

This modal should appear only once , and after clicking on **clear and proceed to setup** button -> user to be able to see the new protocol on the selected robot

### Steps To Reproduce

1. Select a protocol on protocol page
2. Click on the overflow button -> click on run -> choose a robot that already has a protocol loaded -> click on proceed to setup
3. "Robot is busy" modal appears
4. User still wants to use the robot and clear the previous loaded protocol -> clicking on **clear and proceed to setup** button should take me to setup tab
5. But now the modal reappears and vanishes again

https://user-images.githubusercontent.com/39049853/165397100-d912e42f-b741-492b-abd8-ed6f758af40c.mov

nusrat813, 2022-04-28:
👀  from dev needed for next steps

sakibh, 2022-06-16:
Closing this ticket because it was addressed in a separate issue where the robot is busy modal was removed and the protocol will automatically clear and proceed to setup.

closed 2022-06-16 by sakibh