Opentrons/opentrons #10100: 5.1 Feedback:  missing fields (creation method, author, last analyzed , description) on protocol detail page

reachojhaneha, 2022-04-26:
### Overview

Generate a PD protocol with author and description but when you upload that protocol on App -> the details are not populated.

### Current Behavior

[image]

### Expected Behavior

[image]

### Steps To Reproduce

- Upload a protocol on the app
- Goto protocol details page -> creation method, author, description, last analyzed details are not displayed.

reachojhaneha, 2022-04-26:
closing this issue as the feature is still in development. Just verified with Brian about it

closed 2022-04-26 by reachojhaneha

y3rsh, 2022-05-26:
[image]