Opentrons/opentrons #10106: 5.1 Feedback:  Robot Settings > Advanced > Usage settings styling is off

emilywools, 2022-04-27:
### Overview

 Robot Settings > Advanced > Usage settings styling is off
[image]

### Current Behavior

the font size and spacing is not correct for the usage settings section of the advanced app settings

### Expected Behavior

the "usage settings" should be h3-semibold
the individual item title underneath this section should be p-semibold

https://www.figma.com/file/bVtCogdwiXj3ndpk4AWXCx/Milestone-1%3A-Read-and-Execute?node-id=1297%3A90420

### Steps To Reproduce

go to robot > settings > advanced

### Operating system

Mac

koji, 2022-04-27:
@emilywools 
`Usage Settings` is h3 and font-weight 600
[image]

`Pause protocol when robot door opens` is p and font-weight 600
[image]

emilywools, 2022-04-28:
hmm thanks for looking into this @koji. [@mmencarelli](https://github.com/mmencarelli) can you investigate why this looks off?

emilywools, 2022-05-02:
@mmencarelli is going to create a new ticket to capture this feedback. closing for now

closed 2022-05-02 by emilywools